platformio/platformio-vscode-ide #2405: Start PIO Home Server

YiffyToys, 2021-03-03:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.44.2
  PIO IDE: v2.3.0
  System: Windows_NT, 10.0.19042, x64

  %23 Exception
  ```
  Error: Error: Traceback (most recent call last):
  File "C:\Users\Suran\.platformio\penv\lib\site-packages\platformio\__main__.py", line 109, in main
    cli()  %23 pylint: disable=no-value-for-parameter
  File "C:\Users\Suran\.platformio\penv\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\Suran\.platformio\penv\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "C:\Users\Suran\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "C:\Users\Suran\.platformio\penv\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\Suran\.platformio\penv\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\Suran\.platformio\penv\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "C:\Users\Suran\.platformio\penv\lib\site-packages\platformio\commands\home\command.py", line 93, in cli
    run_server(
  File "C:\Users\Suran\.platformio\penv\lib\site-packages\platformio\commands\home\run.py", line 63, in run_server
    contrib_dir = get_core_package_dir("contrib-piohome")
  File "C:\Users\Suran\.platformio\penv\lib\site-packages\platformio\package\manager\core.py", line 48, in get_core_package_dir
    pkg = pm.get_package(spec)
  File "C:\Users\Suran\.platformio\penv\lib\site-packages\platformio\package\manager\base.py", line 236, in get_package
    for pkg in self.get_installed():
  File "C:\Users\Suran\.platformio\penv\lib\site-packages\platformio\package\manager\base.py", line 209, in get_installed
    pkg = PackageItem(pkg_dir)
  File "C:\Users\Suran\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 405, in __init__
    self.metadata = self.load_meta()
  File "C:\Users\Suran\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 440, in load_meta
    return PackageMetaData.load(manifest_path)
  File "C:\Users\Suran\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 391, in load
    data = json.load(fp)
  File "C:\Users\Suran\.platformio\python3\lib\json\__init__.py", line 293, in load
    return loads(fp.read(),
  File "C:\Users\Suran\.platformio\python3\lib\encodings\cp1252.py", line 23, in decode
    return codecs.charmap_decode(input,self.errors,decoding_table)[0]
UnicodeDecodeError: 'charmap' codec can't decode byte 0x81 in position 8: character maps to <undefined>

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at c:\Users\Suran\.vscode\extensions\platformio.platformio-ide-2.3.0\node_modules\platformio-node-helpers\dist\index.js:1:5750
	at ChildProcess.c (c:\Users\Suran\.vscode\extensions\platformio.platformio-ide-2.3.0\node_modules\platformio-node-helpers\dist\index.js:1:930767)
	at ChildProcess.emit (events.js:203:13)
	at ChildProcess.EventEmitter.emit (domain.js:476:20)
	at ChildProcess.cp.emit (c:\Users\Suran\.vscode\extensions\platformio.platformio-ide-2.3.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

YiffyToys, 2021-03-03:
Opened an old project after a year because I urgently needed to change some constants.
Can't build it and thus can't get any work down using the production machinery controlled by this Arduino code
because someone did not catch charset errors and gracefully handled them.

(e.g. making it a warning and displaying source code file and line of the character)
`rocessing megaatmega2560 (platform: atmelavr; board: megaatmega2560; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Error: Traceback (most recent call last):
  File "c:\users\suran\.platformio\penv\lib\site-packages\platformio\__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "c:\users\suran\.platformio\penv\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\suran\.platformio\penv\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "c:\users\suran\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "c:\users\suran\.platformio\penv\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\suran\.platformio\penv\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\suran\.platformio\penv\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "c:\users\suran\.platformio\penv\lib\site-packages\click\decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\suran\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 133, in cli
    process_env(
  File "c:\users\suran\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 175, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "c:\users\suran\.platformio\penv\lib\site-packages\platformio\commands\run\processor.py", line 79, in process
    result = p.run(build_vars, build_targets, self.silent, self.verbose, self.jobs)
  File "c:\users\suran\.platformio\penv\lib\site-packages\platformio\platform\_run.py", line 60, in run
    self.autoinstall_runtime_packages()
  File "c:\users\suran\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 72, in autoinstall_runtime_packages
    if self.get_package(name):
  File "c:\users\suran\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 30, in get_package
    return self.pm.get_package(spec or self.get_package_spec(name))
  File "c:\users\suran\.platformio\penv\lib\site-packages\platformio\package\manager\base.py", line 236, in get_package
    for pkg in self.get_installed():
  File "c:\users\suran\.platformio\penv\lib\site-packages\platformio\package\manager\base.py", line 209, in get_installed
    pkg = PackageItem(pkg_dir)
  File "c:\users\suran\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 405, in __init__
    self.metadata = self.load_meta()
  File "c:\users\suran\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 440, in load_meta
    return PackageMetaData.load(manifest_path)
  File "c:\users\suran\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 391, in load
    data = json.load(fp)
  File "C:\Users\Suran\.platformio\python3\lib\json\__init__.py", line 293, in load
    return loads(fp.read(),
  File "C:\Users\Suran\.platformio\python3\lib\encodings\cp1252.py", line 23, in decode
    return codecs.charmap_decode(input,self.errors,decoding_table)[0]
UnicodeDecodeError: 'charmap' codec can't decode byte 0x81 in position 8: character maps to <undefined>`

YiffyToys, 2021-03-03:
Why the *** does this BUILD ENVIRONMENT even AUTO UPDATE WITHOUT ASKING ME?
It worked fine in 2019 and I did not want to update anything. I just wanted a stable and tested build environment to run once again with no changes to the build environment.

You already did that once with the change to Visual Studio Code and that already cost me a week of downtime.

YiffyToys, 2021-03-03:
Already uninstalled and reinstalled platform-io as told by the error message

YiffyToys, 2021-03-03:
I not only had to uninstall platformIO but also delete %USERHOME%\.platformio .
**So the uninstaller did not properly clean up**.

ivankravets, 2021-03-03:
Could you open [PlatformIO IDE CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) and provide an output from? 
```
pio system info
```

YiffyToys, 2021-03-04:
> 
> 
> Could you open [PlatformIO IDE CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) and provide an output from?
> 
> ```
> pio system info
> ```

There is no context menu on the right mouse button to copy&paste.
As stated above, this is after uninstallung, reinstalling, then uninstalling again because it still didn't work, deleting large amounts of files the uninstaller did not properly remove and reinstalling. 

`pio system info
--------------------------  ------------------------------------------------------
PlatformIO Core             5.1.0
Python                      3.9.1-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\Suran\.platformio
PlatformIO Core Executable  C:\Users\Suran\.platformio\penv\Scripts\platformio.exe
Python Executable           c:\users\suran\.platformio\penv\scripts\python.exe
Global Libraries            0
Development Platforms       1
Tools & Toolchains          4
--------------------------  ------------------------------------------------------
PS C:\Users\Suran\Documents\ProxxonPump_v4_0> `

ivankravets, 2021-03-04:
Does it work now without any issues?

YiffyToys, 2021-03-04:
See 4 comments above.
"After reinstalling it finally worked again."

ivankravets, 2021-03-04:
Yes, it looks you had broken Python installation. Removing `.platformio` folder solved the issue. I see IDE uses the latest portable python 3.9.1

Happy coding with PlatformIO!

YiffyToys, 2021-03-04:
So how does platform.io plan to inform the users about an incompatible python version installed by an earlier version of platform.io ?

ivankravets, 2021-03-04:
Let's see if others report similar issues.

closed 2021-05-11 by stale[bot]