platformio/platformio-vscode-ide #2406: The plugin deletes the core without notice or backup (that I know of)

avihayb, 2021-03-04:
(disclaimer: the issue may be in the core and not the plugin,
it happened to me twice in the last two years, and I couldn't find any issue on github since 18 Mar 2020 (https://github.com/platformio/platformio-core/issues/3425) so it might already be fixed)

The plugin uninstalls the core when it feels like it (and then reinstall an updated version).
How about not uninstalling the core when the computer is offline (no internet connection)?

user story: "I take my old laptop to the field to program a device in the field, and I have 30min of battery left at best, without an internet connection, I press upload, and I'm done"

Current behavior: "I press upload, and I get an error message" 

Possible workarounds:
-try to trigger the update, and suspend the system, then rush to the deployment site
-put the platformIO folders in read-only state after the project compiles (The cost of not being up to date is near zero (in stable cores), so the cost to get to the field twice is infinitely more expensive)
-prey to the FSM and bring spaghetti and meatballs in your field lunchbox just in case

Possible fixes:
-Don't uninstall the core when offline, or better yet
-Don't uninstall the core if the replacement download didn't finish

The last option is a must if you think about it. Imagine what happens when your infrastructure crashes. It will leave a lot of people without the tool they need for work! you'll be condemning people to switch to Arduino "IDE"! a fate worse then death, I assure you (I see that it actually happened: https://github.com/platformio/platformio-core/issues/3319 )

Probably related:
https://github.com/platformio/platformio-core/issues/3333
https://github.com/platformio/platformio-core/issues/3319
https://github.com/platformio/platformio-core/issues/2996
and many more

Triangulix, 2021-04-03:
Seems like we have the same problem. In my case the issue seems to happen every few weeks, maybe exactly once a month. I interpreted this as an intentional function to enforce updates at least once a month.

My ticket: [platformio/platformio-vscode-ide/issues/2398](https://github.com/platformio/platformio-vscode-ide/issues/2398)

rconcept, 2021-04-13:
Hi, 
I've the same problem: 
Every now and then, PlatformIO does an online check while starting VSCode, and without a working internet connection, it deletes the "c:\users\\_YourUsername_\\.platformio\python3" folder and afterwards complains that it cannot find Python3... :man_facepalming:
Very annoying!

One problem is this line:
https://github.com/platformio/platformio-vscode-ide/blob/ad10ea380a04628614d5450cf61e389a83d831c4/src/main.js#L60

With "startInstaller" declared as :
https://github.com/platformio/platformio-vscode-ide/blob/ad10ea380a04628614d5450cf61e389a83d831c4/src/main.js#L152
It seems that it will make an online check (after some days/weeks) if there is at least one project present. So basically, every time.

Some versions before, the same line was:
https://github.com/platformio/platformio-vscode-ide/blob/3a06026e58db3546e49dffe3b348ba1829ebc74b/src/main.js#L61

I tested the old line in the new version by this dirty hack:
**modifying directly the installed file "c:\users\\_YourUsername_\\.vscode\extensions\platformio.platformio-ide-2.3.1\dist\extension.js"**, and searched for :
`,await this.startInstaller(!t),`
and replaced it with :
`,await this.startInstaller(1),`

Everything worked fine. Even putting a system time several months ahead, VSCode started PlatformIO without any problems and I could flash my project to my MCU.

Until pio gets updated, I suggest this hack if you want to be sure that you can continue to use platformIO while offline.
Hopefully this helps you.

rconcept, 2021-04-15:
With commit https://github.com/platformio/platformio-core-installer/commit/33a95e961266a185da8442011accebefce996f0a, this bug (https://github.com/platformio/platformio-vscode-ide/issues/2406) doesn't occur anymore.
A small discrepancy remains: even offline, platformio-vscode-ide is still _trying_ to check for updates, but at least the platformio-core-installer doesn't delete itself anymore if it can't download it.

Triangulix, 2021-06-15:
I guess we get what we pay for.
Fortunately I only need Platformio for the Firmware of my 3D printer. I couldn't imagine using this professionally.

closed 2021-08-21 by stale[bot]