platformio/platformio-vscode-ide #2409: Unable to compile new created project for STM8S board

ernestp, 2021-03-05:
> Executing task in folder 6xS STM Driver: platformio run --environment stm8sblue <

Processing stm8sblue (platform: ststm8; board: stm8sblue; framework: arduino)
---------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm8/stm8sblue.html
PLATFORM: ST STM8 (1.0.3) > ST STM8S103F3 Breakout Board
HARDWARE: STM8S103F3P6 16MHz, 1KB RAM, 8KB Flash
PACKAGES: 
 - framework-arduinoststm8 0.40.181216 
 - tool-stm8binutils 0.230.0 (2.30) 
 - toolchain-sdcc 1.30804.10766 (3.8.4)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 12 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/stm8sblue/src/main.cpp.rel
Archiving .pio/build/stm8sblue/libFrameworkArduinoVariant.lib
Indexing .pio/build/stm8sblue/libFrameworkArduinoVariant.lib
Compiling .pio/build/stm8sblue/FrameworkArduino/HardwareSerial.c.rel
Compiling .pio/build/stm8sblue/FrameworkArduino/Print-float.c.rel
Compiling .pio/build/stm8sblue/FrameworkArduino/Print.c.rel
In file included from src/main.cpp:1:
In file included from /Users/user/.platformio/packages/framework-arduinoststm8/cores/sduino/Arduino.h:23:
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/stdlib.h:97:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/stdlib.h:168:1: error: 'long float' is invalid
long double
^
In file included from src/main.cpp:1:
In file included from /Users/user/.platformio/packages/framework-arduinoststm8/cores/sduino/Arduino.h:23:
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/stdlib.h:100:
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/math.h:303:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/math.h:133:26: error: 'long float' is invalid
extern int __fpclassifyl(long double);
                         ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/math.h:176:47: error: 'long float' is invalid
__header_always_inline int __inline_isfinitel(long double);
                                              ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/math.h:179:44: error: 'long float' is invalid
__header_always_inline int __inline_isinfl(long double);
                                           ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/math.h:182:44: error: 'long float' is invalid
__header_always_inline int __inline_isnanl(long double);
                                           ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/math.h:185:47: error: 'long float' is invalid
__header_always_inline int __inline_isnormall(long double);
                                              ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/math.h:188:46: error: 'long float' is invalid
__header_always_inline int __inline_signbitl(long double);
                                             ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/math.h:196:47: error: 'long float' is invalid
__header_always_inline int __inline_isfinitel(long double __x) {
                                              ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/math.h:205:44: error: 'long float' is invalid
__header_always_inline int __inline_isinfl(long double __x) {
                                           ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/math.h:214:44: error: 'long float' is invalid
__header_always_inline int __inline_isnanl(long double __x) {
                                           ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/math.h:228:46: error: 'long float' is invalid
__header_always_inline int __inline_signbitl(long double __x) {
                                             ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/math.h:230:9: error: 'long float' is invalid
        long double __ld;
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/math.h:249:47: error: 'long float' is invalid
__header_always_inline int __inline_isnormall(long double __x) {
                                              ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/math.h:310:8: error: 'long float' is invalid
extern long double acosl(long double);
       ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/math.h:310:26: error: 'long float' is invalid
extern long double acosl(long double);
                         ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/math.h:314:8: error: 'long float' is invalid
extern long double asinl(long double);
       ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/math.h:314:26: error: 'long float' is invalid
extern long double asinl(long double);
                         ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/math.h:318:8: error: 'long float' is invalid
extern long double atanl(long double);
       ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/math.h:318:26: error: 'long float' is invalid
extern long double atanl(long double);
                         ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
/Users/user/.platformio/packages/framework-arduinoststm8/cores/sduino/HardwareSerial.c:38:2: warning: #warning "using uart1 for HardwareSerial"
Compiling .pio/build/stm8sblue/FrameworkArduino/SPI.c.rel
Compiling .pio/build/stm8sblue/FrameworkArduino/WInterrupts.c.rel
Compiling .pio/build/stm8sblue/FrameworkArduino/WMath.c.rel
Compiling .pio/build/stm8sblue/FrameworkArduino/weak_atexit.c.rel
20 errors generated.
*** [.pio/build/stm8sblue/src/main.cpp.rel] Error 1
/Users/user/.platformio/packages/framework-arduinoststm8/cores/sduino/weak_atexit.c:13: warning 85: in function atexit unreferenced function argument : 'func'
/Users/user/.platformio/packages/framework-arduinoststm8/cores/sduino/WInterrupts.c:73: warning 85: in function attachInterrupt unreferenced function argument : 'mode'
========================================= [FAILED] Took 1.04 seconds =========================================
The terminal process "platformio 'run', '--environment', 'stm8sblue'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

ivankravets, 2021-03-05:
Please report to https://community.platformio.org/

closed 2021-03-05 by ivankravets

ernestp, 2021-03-06:
Moved to https://community.platformio.org/t/unable-to-compile-new-created-project-for-stm8s-board/19766