platformio/platformio-vscode-ide #2410: Having to separately select a default to compile and upload since v2.3.0

Cathprotech, 2021-03-05:
Since v2.3.0 there is now requirement to select separately which project to build and upload even though this can be different to the actual project being worked on.
Caused confusion and resulted in OTA wiping of code / programming incorrect code to a device that was not being worked on resulting, in my case a loss of central heating control due to the option being (unknown to me) selected for my central heating programmer rather than the project I was working on.
This change could have been made more prominent.
If it wasn't for asking on the support forum I would have had to revert to using Arduino IDE to re-program my device after first physically removing it from its housing, as I would not know any other way to re-program or program that or any other device.
Can this be made optional to either be able to select the 'default' project or have the previous option of automatic selection to the project currently being worked on as per the V2.2.1 release?

ivankravets, 2021-03-05:
Thanks! We will think about adding a setting to have previous behavior that brought a ton of issues. People preferred declarative use of this extension instead of jumping between multiple projects depending on opened file in a text editor.

key1066, 2021-03-09:
Thanks Ivan and Cathprotech.

There is no doubt that the change in v2.3.0 to select a ‘default’ project for build & upload will fix some of the quirks of the v2.2.1 behaviour of switching to the env/project to which the currently open source file belongs. However, for those of us who routinely work on multiple projects and devices, the ability to automatically switch to a currently open source file and build the project & move between projects in the manner of v2.2.1 is more productive and faster despite the quirks.

An option to select between the current behaviour of v2.3.0 and the previous v2.2.1 would be good. 

Perhaps a quicker method of switching to a ‘default’ project, for example, a right mouse click on a project folder in the workspace tree view to make a given project the  ‘default’ project could provide the best both worlds of current and v2.2.1.

emielsteerneman, 2021-03-12:
I'm trying to figure out how to change the default projects, but it is unclear to me. For now I have to resort to closing all projects but the one I want to build. How does changing default projects work?

cad435, 2021-03-14:
> I'm trying to figure out how to change the default projects, but it is unclear to me. For now I have to resort to closing all projects but the one I want to build. How does changing default projects work?

[image]
For me its down there (PIO on VSCode)

key1066, 2021-03-14:
You can make a project `Default` as follows.

1. Click on the `Default` icon at the bottom menu bar, then select a project from the `Default/Env` list.

2. You have the following options from the `Defaults/Env` list
- Default `(“Default_envs” from “platform.ini”)`
- If you have set up additional `env` entries for other boards, you’ll see  `env:<board>` options to select from as your default project.

It would be handy to have an option to highlight a project in the project workspace folder tree, right-mouse click and then select an option from the pop-up menu  to make the selected project the `Default`.

Bascy, 2021-03-15:
See also #2408

closed 2021-03-23 by ivankravets

ivankravets, 2021-03-23:
PlatformIO IDE v2.3.1 for VSCode is out! Please update https://github.com/platformio/platformio-vscode-ide/releases/tag/v2.3.1

Cathprotech, 2021-03-23:
Thank you for the update but I don't see any difference.
I still have to select the project I am working on as default. Am I missing something?
Thanks
Martin

ivankravets, 2021-03-24:
Please check how to use VSCode settings: https://docs.platformio.org/en/latest/integration/ide/vscode.html#settings

- https://marketplace.visualstudio.com/items/platformio.platformio-ide/changelog
- https://github.com/platformio/platformio-vscode-ide/releases/tag/v2.3.1

You need to enable `platformio-ide.activateProjectOnTextEditorChange`

Cathprotech, 2021-03-24:
Brilliant
Thank you

Galaxy-Man, 2021-04-21:
Thank you for this, it solved an issue I had for ages. Works well now :-)

Galaxy-Man, 2021-04-22:
I spoke too soon :-( 

I am not one to give up but I have spent yesterday and this morning trying to understand VSC PlatformIO 2.3.1 and why I still have an issue with an open project building a different main.cpp!

Default is set to the main.cpp that is open, no other project is open yet it builds a different one.
I have a workspace with 20 projects stored in there eah with different platformio.ini files in correct place. 

For my workspace i have set the following
platformio-ide.activateProjectOnTextEditorChange = true
platformio-ide.activateOnlyOnPlatformIOProject = true
platformio-ide.autoOpenPlatformIOIniFile = true

Anyone pleases help as I am pulling my hair out now and I don't have much to start with :-)

Galaxy-Man, 2021-04-22:
[image]

Galaxy-Man, 2021-04-22:
[image]

ivankravets, 2021-04-24:
COuld you DOUBLE CLICK on main.cpp? It is not activated in VSCode editor.

Galaxy-Man, 2021-04-25:
The tab is now not italic. I will give it a go hopefully this sill fix it.

Is there a link to webpage describing this as I could not find it.

Thnaks for your help.

paulskirk53, 2022-01-17:
brilliant feature which has saved me having to remember that which cannot be reliably remembered...

DragonflyNet67, 2025-02-01:
> Please check how to use VSCode settings: https://docs.platformio.org/en/latest/integration/ide/vscode.html#settings
>
>     * [...]

Great, exactly what i was looking for. Thanks

commit 577e5216 (2021-03-23): Added a new setting platformio-ide.activateProjectOnTextEditorChange to enable automatic project activation depending on an active opened text editor // Resolve #2410