platformio/platformio-vscode-ide #2412: Add a "Changelog" page on extension updates

maxgerhardt, 2021-03-06:
The change in the project creation and selection workflow through the project and environment selector in extension version v2.3.0, has left many users baffled and frusted, as in e.g.

* https://community.platformio.org/t/ide-compiles-wrong-project-file/19399
* https://community.platformio.org/t/platformio-compiles-wrong-projects/19769
* https://community.platformio.org/t/platfromio-not-working/19763/5
* https://community.platformio.org/t/ota-uploads-to-wrong-address/19745/11

4 topics created yesterday and today since the release the day before yesterday.

Of course #2410 is related to that in order to be able to have the old behavior (which was reported buggy by users), but I also think it's important that when these workflow changes occurr, that the extension opens up a page after an update where the critical changes are explained.

maxgerhardt, 2021-03-14:
This occurred again in https://community.platformio.org/t/platformio-tries-to-upload-to-wrong-port-even-if-declared-in-platformio-ini/19989 -- if a user hasn't worked with PlatformIO for a while and suddenly the workflow changes, confusion is bound to happen.

ivankravets, 2022-12-26:
Hi @maxgerhardt ,

Please re-test with the pre-release version. See the instructions below:

[image]

Have a great holiday season! 🤗 🎄

commit e0c04b3d (2022-12-26): Provide PlatformIO IDE Release Notes // Resolve #2412
closed 2022-12-26 by ivankravets