platformio/platformio-vscode-ide #2413: Cannot install PIO

JindrazZ, 2021-03-06:
Hi there. I am not programmer. I just want to tweak something in marlin via VScode and I follow a tutorial on ytb. 
I am having this issue with instaling PIO. I tried to reinstall it several times, with different versions etc., even on Atom and its a no go. 
I have looked at other people issues but didnt found anything that would help me... 
Any help please ? 

Error: Error: Traceback (most recent call last):
  File "c:\.platformio\penv\lib\site-packages\platformio\__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "c:\.platformio\penv\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "c:\.platformio\penv\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "c:\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "c:\.platformio\penv\lib\site-packages\click\core.py", line 1254, in invoke
    cmd_name, cmd, args = self.resolve_command(ctx, args)
  File "c:\.platformio\penv\lib\site-packages\click\core.py", line 1297, in resolve_command
    cmd = self.get_command(ctx, cmd_name)
  File "c:\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 64, in get_command
    mod = __import__(mod_path, None, None, ["cli"])
  File "c:\.platformio\penv\lib\site-packages\platformio\commands\home\command.py", line 19, in <module>
    from platformio.commands.home.helpers import is_port_used
  File "c:\.platformio\penv\lib\site-packages\platformio\commands\home\helpers.py", line 26
    async def request(  # pylint: disable=signature-differs,invalid-overridden-method
            ^
SyntaxError: invalid syntax

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at c:\Users\jindra\.vscode\extensions\platformio.platformio-ide-1.8.3\node_modules\platformio-node-helpers\dist\index.js:1:17279
	at ChildProcess.h (c:\Users\jindra\.vscode\extensions\platformio.platformio-ide-1.8.3\node_modules\platformio-node-helpers\dist\index.js:1:6057)
	at ChildProcess.emit (events.js:315:20)
	at ChildProcess.cp.emit (c:\Users\jindra\.vscode\extensions\platformio.platformio-ide-1.8.3\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)

maxgerhardt, 2021-03-06:
> async def request( # pylint: disable=signature-differs,invalid-overridden-method
^
SyntaxError: invalid syntax

What Python version are you using so that it doesn't know `async`?

JindrazZ, 2021-03-06:
I had 3.6 so i tried 3.8 and 3.9 now and doesnt install. Error now with python  v3.8 installed is :

Error: Traceback (most recent call last):
  File "C:\.platformio\.cache\tmp\.piocore-installer-44_ucq46\tmpa1lihebl\pioinstaller.zip\pioinstaller\python.py", line 170, in find_compatible_pythons
  File "C:\.platformio\python3\lib\subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\.platformio\python3\lib\subprocess.py", line 524, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['C:\\Users\\jindra\\AppData\\Local\\Microsoft\\WindowsApps\\python3.exe', 'C:\\.platformio\\.cache\\tmp\\get-platformio-1.0.0.py', '--no-shutdown-piohome', 'check', 'python']' returned non-zer...

maxgerhardt, 2021-03-06:
>C:\Users\jindra\AppData\Local\Microsoft\WindowsApps\python3.exe

I've seen problems with the microsoft-provided Python installation multiple times. Can you check to make sure you rearrange the order of the paths in the system `PATH` variable (https://www.computerhope.com/issues/ch000549.htm) so that the Python installation in like `C:\Users\<user>\AppData\Local\Programs\Python\Python38\` is first? Also refer #2411

JindrazZ, 2021-03-06:
Well I installed python myself from web, so could it be that there is windows provided copy on my PC too ? Once (among like 15 attemepts) during instalation PIO asked for folder of python, but then failed and never asked again...

" Can you check to make sure you rearrange the order of the paths in the system PATH variable "
is this what you mean ? this is from user tab 
[image]
and this one from system
[image]

maxgerhardt, 2021-03-06:
Hm yes it's in the user's variables. I see Python3.8 (in the 32-bit version weirdly) is already on top of the Microsoft one that's stored in WindowsApps. 

Have you restarted your computer to check that there are some pending changes in the PATH that mess things up here? Can you check if it helps to remove the `WindowsApps` folder from the PATH?

JindrazZ, 2021-03-06:
"remove the WindowsApps folder "
Aah, yes this finally worked! Thank you for the time you spend on helping.
Now PATH look like this and it allowed to install PIO.
[image]

closed 2021-03-06 by JindrazZ

maxgerhardt, 2021-03-06:
Hmm that may work for now but maybe not having the WindowsApps in the Path is critical for other applications. For me, there's also Python3 in there along with Spotify and Skype and the Windows terminal app..

[image]

My path looks like https://community.platformio.org/t/i-cant-install-platfomio-in-the-last-version-2-3-0/19776/5?u=maxgerhardt (Python 3.8 first and then Microsoft's version) my PlatformIO installation works.