platformio/platformio-vscode-ide #2414: Automatically switch to newly created project's environment

maxgerhardt, 2021-03-06:
When creating a new project through the PIO Home "Create Project" dialogue, the project and environment selection is not automatically updated to the new project. I think this would be a good automatic thing to do since when a user creates a new project, he probably wants to work on it immediately and not in the old project. Maybe also a short popup to notify the user of that would be nice.

pfeerick, 2021-03-07:
Agreed. I had noticed this breakage the other day and thought it was due to a VSCode update, not a change in the PIO Extension... but now I see what's happening.

A further request would be to re-introduce the old behaviour, if possible, of switching to the environment/project to which the currently open source file belongs. I often have a workspace with two or three projects, and quite liked that when I switched between the main.cpp tabs for two (i.e. a one project would Tx, the other would Rx) I could build and upload via the statusbar. Now that doesn't happen as it won't auto-switch. 

I believe I was also suffering from some of the issues this was seeking to fix, as selecting the upload and build tasks a couple days ago on my linux box in the project tasks sidebar was building/uploading the wrong envs, so hopefully that is gone now.

ivankravets, 2021-03-23:
PlatformIO IDE v2.3.1 for VSCode is out! Please update https://github.com/platformio/platformio-vscode-ide/releases/tag/v2.3.1

maxgerhardt, 2021-03-30:
I've tested this in 2.3.1 but I don't see this new behavior.

Example: 3 projects are in the workspace. I first switch to the lowest project in the list. The environment status bar shows that.

[image]

I go to 'New Project' and create a new project.

[image]

I press finish and..

[image]

A new folder is added, but it isn't automatically switched to. It also doesn't auto-open the `platformio.ini`.

```
Version: 1.54.3 (system setup)
Commit: 2b9aebd5354a3629c3aba0a5f5df49f43d6689f8
Date: 2021-03-15T10:55:45.459Z
Electron: 11.3.0
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Windows_NT x64 10.0.19042
```

[image]

Debug Log 

[image]

commit 0a5ae4ed (2021-03-22): Automatically activate project environment opened via PIO Home > New Project / Open Project // Resolve #2414
commit 95d7f28a (2021-11-04): Automatically switch to newly created project's environment // Resolve #2414
closed 2021-11-04 by ivankravets