platformio/platformio-vscode-ide #2417: Start PIO Home Server

claucrivel, 2021-03-07:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.54.1
  PIO IDE: v2.3.0
  System: Windows_NT, 10.0.18363, x64

  %23 Exception
  ```
  Error: Could not start PIO Home server: Error: connect EADDRNOTAVAIL 127.0.0.1 - Local (0.0.0.0:57666)
	at c:\Users\clacr\.vscode\extensions\platformio.platformio-ide-2.3.0\node_modules\platformio-node-helpers\dist\index.js:1:5828
	at runMicrotasks (<anonymous>)
	at processTicksAndRejections (internal/process/task_queues.js:97:5)
  ```

ivankravets, 2021-03-20:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/2451

ivankravets, 2021-03-20:
Do you use 3rd party antivirus tools?

closed 2021-03-20 by ivankravets

claucrivel, 2021-03-22:
Hi Ivan, yes it is. I use McAfee antivirus

Enviada em: sábado, 20 de março de 2021 15:51
Para: platformio/platformio-vscode-ide ***@***.***>
Cc: claucrivel ***@***.***>; Author ***@***.***>

ivankravets, 2021-03-22:
It seems McAfee blocks PlatformIO on starting PIO Home server.