platformio/platformio-vscode-ide #2420: Unable to install pio core in VSC 1.54.1

iftahnaf, 2021-03-09:
Hi all,

I'm running VSC with PIO extension on Ubuntu 18.04

A few days ago my VSC update itself to 1.54.1, and from then I can't load/reinstall PIO, It's stuck in the middle of the installation process (before the update it worked fine)

In VSC Help-->Toggle Dev. Tools --> Console I get the following error:

`Error:   Cache entry deserialization failed, entry ignored   Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='pypi.org', port=443): Read timed out. (read timeout=15)",)': /simple/platformio/   Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='pypi.org', port=443): Read timed out. (read timeout=15)",)': /simple/platformio/   Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='pypi.org', port=443)...`

And this:

```
RequestError: connect ECONNREFUSED 127.0.0.1:8049
	at ClientRequest.<anonymous> (/home/iftach/.vscode/extensions/platformio.platformio-ide-2.3.0/node_modules/got/dist/source/core/index.js:956:111)
	at Object.onceWrapper (events.js:422:26)
	at ClientRequest.emit (events.js:327:22)
	at ClientRequest.EventEmitter.emit (domain.js:483:12)
	at ClientRequest.origin.emit (/home/iftach/.vscode/extensions/platformio.platformio-ide-2.3.0/node_modules/@szmarczak/http-timer/dist/source/index.js:39:20)
	at Socket.socketErrorListener (_http_client.js:426:9)
	at Socket.emit (events.js:315:20)
	at Socket.EventEmitter.emit (domain.js:483:12)
	at emitErrorNT (internal/streams/destroy.js:92:8)
	at emitErrorAndCloseNT (internal/streams/destroy.js:60:3)
	at processTicksAndRejections (internal/process/task_queues.js:84:21)
	at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1141:16)
DevTools failed to load SourceMap: Could not load content for https://ticino.blob.core.windows.net/sourcemaps/f30a9b7…/core/bootstrap-window.js.map:
```

Any way to solve this problem?
Thanks!

iftahnaf, 2021-03-10:
> Do you use a proxy? Could you check if you have `HTTP_PROXY` or `HTTPS_PROXY` environment variables?

Thanks for the answer.

I'm not using proxy and there are no environment variables `HTTP_PROXY` or `HTTPS_PROXY`

ivankravets, 2021-03-13:
Can you access pypi.org from your browser?

iftahnaf, 2021-03-13:
Yes, I could access pypi.org

Suddenly after a few days, the error gone and the PIO extension works fine now.

Thanks again.

closed 2021-03-13 by iftahnaf