platformio/platformio-vscode-ide #2434: Start PIO Home Server

smougenot, 2021-03-13:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.54.2
  PIO IDE: v2.3.0
  System: Linux, 5.10.21-100.fc32.x86_64, x64

  %23 Exception
  ```
  Error: Usage: __main__.py home [OPTIONS]
Try '__main__.py home -h' for help.

Error: no such option: --session-id

	at /home/smougenot/.vscode/extensions/platformio.platformio-ide-2.3.0/node_modules/platformio-node-helpers/dist/index.js:1:5750
	at ChildProcess.c (/home/smougenot/.vscode/extensions/platformio.platformio-ide-2.3.0/node_modules/platformio-node-helpers/dist/index.js:1:930767)
	at ChildProcess.emit (events.js:315:20)
	at ChildProcess.EventEmitter.emit (domain.js:483:12)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)
  ```

ivankravets, 2021-03-13:
Please remove https://docs.platformio.org/en/latest/faq.html#multiple-platformio-cores-in-a-system and restart VSCode.

You have outdated PlatformIO Core in a system.

closed 2021-03-13 by ivankravets

smougenot, 2021-03-14:
@ivankravets sorry to bother you about your answer. 
I'd loose pio shell integration because it is not any more in `~/.platformio/penv/bin/`
So :
* is [the doc](https://docs.platformio.org/en/latest/core/installation.html#unix-and-unix-like) up to date?
* is shell integration abandoned if we use the integration with VSCode (`PlatformIO IDE `)?
* or maybe there is a glitch in the last update

```
smougenot  ~  ll ~/.platformio/penv/bin/
total 56
drwxr-xr-x. 2 smougenot smougenot 4096 Mar 13 09:22 .
drwxr-xr-x. 5 smougenot smougenot 4096 Mar 13 09:22 ..
-rw-r--r--. 1 smougenot smougenot 2191 Mar 13 09:22 activate
-rw-r--r--. 1 smougenot smougenot 1243 Mar 13 09:22 activate.csh
-rw-r--r--. 1 smougenot smougenot 2395 Mar 13 09:22 activate.fish
-rw-r--r--. 1 smougenot smougenot 8834 Mar 13 09:22 Activate.ps1
-rwxr-xr-x. 1 smougenot smougenot  258 Mar 13 09:21 easy_install
-rwxr-xr-x. 1 smougenot smougenot  258 Mar 13 09:21 easy_install-3.8
-rwxr-xr-x. 1 smougenot smougenot  248 Mar 13 09:22 pip
-rwxr-xr-x. 1 smougenot smougenot  248 Mar 13 09:22 pip3
-rwxr-xr-x. 1 smougenot smougenot  248 Mar 13 09:22 pip3.8
lrwxrwxrwx. 1 smougenot smougenot    7 Mar 13 09:21 python -> python3
lrwxrwxrwx. 1 smougenot smougenot   16 Mar 13 09:21 python3 -> /usr/bin/python3
-rwxr-xr-x. 1 smougenot smougenot  235 Mar 13 09:22 wheel
```

ivankravets, 2021-03-14:
Please check that you have enabled https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-ide-usebuiltinpiocore

1. Remove `~/.platformio/penv`
2. Restart VSCode.

Shortly, you broke your configuration.

smougenot, 2021-03-17:
Very frustrating, I tried to  help with the error suggested by the PIO Home and now my workstation is screwed
* no more pio
* extensions fails to install in VSCode
* lost hours looking for a solution ... without success

smougenot, 2021-03-17:
sample error from the VScode console
```
/usr/share/code/resources/app/out/vs/workbench/workbench.desktop.main.js:62 [Extension Host] stack trace: RequestError: connect ECONNREFUSED 127.0.0.1:8012	at ClientRequest.<anonymous> (/home/smougenot/.vscode/extensions/platformio.platformio-ide-2.3.0/node_modules/got/dist/source/core/index.js:956:111)	at Object.onceWrapper (events.js:422:26)	at ClientRequest.emit (events.js:327:22)	at ClientRequest.EventEmitter.emit (domain.js:483:12)	at ClientRequest.origin.emit (/home/smougenot/.vscode/extensions/platformio.platformio-ide-2.3.0/node_modules/@szmarczak/http-timer/dist/source/index.js:39:20)	at Socket.socketErrorListener (_http_client.js:426:9)	at Socket.emit (events.js:315:20)	at Socket.EventEmitter.emit (domain.js:483:12)	at emitErrorNT (internal/streams/destroy.js:92:8)	at emitErrorAndCloseNT (internal/streams/destroy.js:60:3)	at processTicksAndRejections (internal/process/task_queues.js:84:21)	at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1141:16)
```

smougenot, 2021-03-17:
Uninstalling VScode + removing vscode files + removing plateformio files (from home folder) did  not solve the problem
Process stuck at plaetformio core installation
```
Installer version: 0.3.5
Platform: Linux-5.10.21-100.fc32.x86_64-x86_64-with-glibc2.2.5
Python version: 3.8.7 (default, Jan 20 2021, 00:00:00) 
...
Updating Python package manager (PIP) in a virtual environment
```

smougenot, 2021-03-17:
End of investigation for me
Will use pio core with cli only

ivankravets, 2021-03-17:
Could you open the system terminal and run the installer script directly? https://docs.platformio.org/en/latest/core/installation.html#super-quick-mac-linux