platformio/platformio-vscode-ide #2437: PlatformIO Extension Doesn't Install Properly

mahmoodfathy, 2021-03-15:
Description of the issue :
Platform io doesnt install properly on windows 10 , this is my current setup:

Version: 1.54.2 (user setup)
Commit: fd6f3bce6709b121a895d042d343d71f317d74e7
Date: 2021-03-11T00:56:19.848Z
Electron: 11.3.0
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Windows_NT x64 10.0.18363

after installing the icon of platformio didn't show up besides loading my cpu usage up to 50% which is pretty unusual , upon opening the developer tools in vs code i get this error message :
WARN UNRESPONSIVE extension host, 'platformio.platformio-ide' took 99% of 5058.134ms, saved PROFILE here: 'file:///c%3A/Users/mahmo/AppData/Local/Temp/exthost-942462.cpuprofile' 
(2) [{…}, {…}]
0: {id: "gc", total: 26500, percentage: 1}
1: {id: "platformio.platformio-ide", total: 5031519, percentage: 99}
length: 2
__proto__: Array(0)

i tried uninstalling and installing python ,removing the extension and folders in .vscode multiple times but nothing seems to work 
any help would be appreciated

ivankravets, 2021-03-20:
Do you use 3rd party antivirus tools in a system?

mahmoodfathy, 2021-03-20:
@ivankravets  no i don't use any 3rd party antivirus tools , it's weird cause it installed no problems whatsoever on my linux machine , i tried all solutions proposed on forums and close issues here , it doesn't seem to work at all , and the wierd part is my cpu usage , it goes up like crazy  once it finished installing, when i removed it  , cpu usage goes back to normal

ivankravets, 2021-03-21:
1. Please uninstall all Python interpreters from your Windows machine
2. PlatformIO IDE comes with own pre-built portable Python, it will be used automatically
3. Remove `%USERPROFILE%/.platformio` folder.
4. Restart VSCode. 
5. Open VSCode > Menu > Help > Toggle Developer Tools > Console. Share the last messages here.

closed 2021-05-27 by stale[bot]