platformio/platformio-vscode-ide #2445: Start PIO Home Server

Vcelist, 2021-03-18:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue

  %23 Configuration

  VSCode: 1.54.3
  PIO IDE: v2.2.1
  System: Windows_NT, 6.1.7601, ia32

  %23 Exception
  ```
  Error: Could not start PIO Home server: Error: connect EADDRNOTAVAIL 127.0.0.1 - Local (0.0.0.0:9844)
	at c:\Users\V\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\platformio-node-helpers\dist\index.js:1:5703
	at runMicrotasks (<anonymous>)
	at processTicksAndRejections (internal/process/task_queues.js:97:5)
  ```

ivankravets, 2021-03-20:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/2451

ivankravets, 2021-03-20:
Do you use 3rd party antivirus tools?

closed 2021-03-20 by ivankravets

Vcelist, 2021-03-22:
No, and I'm disable firewall

Vcelist, 2021-03-22:
and PIO don't work with installed python - it ask to install it

Vcelist, 2021-03-22:
i'm clik on "i have pyton" and paste full way to python.exe but it  "PlatformIO: Can not find working Python 2.7 or 3.5+ Interpreter. Please install the latest Python 3 and restart VSCode"