platformio/platformio-vscode-ide #2451: Start PIO Home Server

Jynxter, 2021-03-20:
%23 Description of problem
  No idea what I'm doing.... It asked to submit a bug report, so I installed Python, PlatformIO, and restarted VSCode. After opening a Configuration.h for a marlin firmware, I got this error. Going to restart everything and try again, but figured any bug report is a good bug report. Note: Search for Dups link below doesn't work.

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.51.1
  PIO IDE: v2.3.0
  System: Windows_NT, 10.0.18363, x64

  %23 Exception
  ```
  Error: Could not start PIO Home server: Error: connect EADDRNOTAVAIL 127.0.0.1 - Local (0.0.0.0:65157)
	at c:\Users\Paul\.vscode\extensions\platformio.platformio-ide-2.3.0\node_modules\platformio-node-helpers\dist\index.js:1:5828
	at processTicksAndRejections (internal/process/task_queues.js:94:5)
  ```

ivankravets, 2021-03-20:
Do you use 3rd party antivirus tools?

ivankravets, 2021-03-20:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/148

closed 2021-03-20 by ivankravets