platformio/platformio-vscode-ide #2467: Build from taskbar starts building wrong project (again)

Bascy, 2021-03-25:
I have multiple projects in a VSCode workspace, I set the settings "Activate project on Text Editor change" enabled, and the active project mentioned in the taskbar does actually change when opening a file from a different project.

But when i press the build (or build & upload) icon in the taskbar, PlatformIO starts building the wrong project, namely the project I previously build. If I open a sourcefile from a different project, which is based on a different board, then it does work correctly.

If I manually choose a project and environment to builk by clicking on the "Switch PlatformIO" taskbar item, then it still builds the wrong porject

[image]

ivankravets, 2021-03-27:
I can't reproduce this issue. Could you open VSCode > Help > Toggle Developers Tools > Console? Try to switch between projects. You will see all info in the logs. Is it correct?

Bascy, 2021-03-28:
I can't reproduce it anymore either :(

I do get an error once in the console, but don't thin it is related to this 

[image]

ivankravets, 2021-04-01:
That error is OK, it comes from VSCode. I'm going to close this issue. If you will experience this problem, please ping me here.

closed 2021-04-01 by ivankravets