platformio/platformio-vscode-ide #2469: Start PIO Home Server

syntekker, 2021-03-26:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.52.1
  PIO IDE: v2.2.1
  System: Linux, 4.4.0-104-generic, x64

  %23 Exception
  ```
  Error: Error: Traceback (most recent call last):
  File "/home/brett/.platformio/penv/lib/python3.5/site-packages/platformio/__main__.py", line 109, in main
    cli()  %23 pylint: disable=no-value-for-parameter
  File "/home/brett/.platformio/penv/lib/python3.5/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/brett/.platformio/penv/lib/python3.5/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/brett/.platformio/penv/lib/python3.5/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/home/brett/.platformio/penv/lib/python3.5/site-packages/click/core.py", line 1254, in invoke
    cmd_name, cmd, args = self.resolve_command(ctx, args)
  File "/home/brett/.platformio/penv/lib/python3.5/site-packages/click/core.py", line 1297, in resolve_command
    cmd = self.get_command(ctx, cmd_name)
  File "/home/brett/.platformio/penv/lib/python3.5/site-packages/platformio/commands/__init__.py", line 64, in get_command
    mod = __import__(mod_path, None, None, ["cli"])
  File "/home/brett/.platformio/penv/lib/python3.5/site-packages/platformio/commands/home/command.py", line 19, in <module>
    from platformio.commands.home.helpers import is_port_used
  File "/home/brett/.platformio/penv/lib/python3.5/site-packages/platformio/commands/home/helpers.py", line 18, in <module>
    from starlette.concurrency import run_in_threadpool
  File "/home/brett/.platformio/penv/lib/python3.5/site-packages/starlette/concurrency.py", line 60
    yield await run_in_threadpool(_next, iterator)
               ^
SyntaxError: 'yield' inside async function

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at /home/brett/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:5625
	at ChildProcess.c (/home/brett/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:879169)
	at ChildProcess.emit (events.js:223:5)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

ivankravets, 2021-03-27:
> PIO IDE: v2.2.1

Please upgrade the extension to the latest version. Restart VSCode.

closed 2021-03-27 by ivankravets

syntekker, 2021-03-27:
Upgrading to the latest version it did not help. The solution was to
install PlatformIO 2.2.1 and Arduino 0.3.5 into the latest VS Code.
I'm guessing that my Ubuntu install Python version is causing issues with
the latest releases.
Whatever the problem it does not happen with this setup and everything is
working as expected.
Thanks for the reply. First time with VS Code and very impressed.