platformio/platformio-vscode-ide #2472: i cant open paltformio ide

ofek195, 2021-03-27:
Error: Could not start PIO Home server: Error: timeout
	at c:\Users\ofek0\.vscode\extensions\platformio.platformio-ide-2.3.0\node_modules\platformio-node-helpers\dist\index.js:1:5828...

ofek195, 2021-03-27:
Error: WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='pypi.org', port=443): Read timed out. (read timeout=15)")': /simple/platformio/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='pypi.org', port=443): Read timed out. (read timeout=15)")': /simple/platformio/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='pypi.org', port=443): Read timed out. (read timeout...

ivankravets, 2021-03-27:
> HTTPSConnectionPool(host='pypi.org', port=443): Read tim

Please check your Internet connection

closed 2021-03-27 by ivankravets