platformio/platformio-vscode-ide #2479: Platfomaio IDE nao inicializa baixa todos os arquivos mas nap inicializa

AlemarFernandes, 2021-03-30:
Este e o print do que eu vejo quando abro nao sei o que esta aconetecndo ....
Sou iniciante mas tenho muito desejo de aprender ...

[image]

CNflysky, 2021-03-30:
i have the same problem as yours.last week i had my platformio ide upgraded and this error shown up,after looking at the issues on github,i erased all files related to platformio ide on my computer and reinstalled it and finally it works well,but today i have it upgraded again and i can't open the mainpage .i hope the developers will solve this problem
p.s. forgive my pool english
[image]

AlemarFernandes, 2021-03-30:
e uma pena sou iniciante , quando isso apareceu fiquei muito triste e frustrado mas vou tentando se ver alguma coisa eu aviso 

sou do brasileiro me descupe pelo ingles uso o tradutor ....l

closed 2021-06-06 by stale[bot]