platformio/platformio-vscode-ide #2481: Investigate incompatibility with Google Cloud Code extension

maxgerhardt, 2021-03-30:
Reported by a user per https://community.platformio.org/t/esp-prog-debugger-issue-no-visible-error-ubuntu/20341/42?u=maxgerhardt, PlatformIO debugging does not operate correctly anymore (quickly terminates) if the Cloud Code extension (presumably https://marketplace.visualstudio.com/items?itemName=GoogleCloudTools.cloudcode) is active.

It should be investigated on why that is and put it on the known-incompatible list or solve the compatibility issue.

supcik, 2021-04-22:
I am having this issue too. Thank you for your investigation.

maxgerhardt, 2021-06-21:
*Bump.* this is still occuring for people at e.g. [here](https://community.platformio.org/t/im-struggling-to-get-started-with-esp-prog-debugging/22090/5?u=maxgerhardt) and is not resolved.

maxgerhardt, 2021-08-21:
@ivankravets please add the Google Cloud Code extension into 

https://github.com/platformio/platformio-vscode-ide/blob/d448a233a1b328d794db35272b6098cf3ad75ec1/src/constants.js#L14-L18

maxgerhardt, 2021-10-13:
@supcik can you confirm the extension is not an issue anymore (possible update) or provide a procedure to reproduce the problem?