platformio/platformio-vscode-ide #2506: Slow "PlatformIO: Rebuilding IntelliSense Index" only first time when reloading VScode after installing PIO

GunZi200, 2021-04-11:
It can take up to 1 minute and 30 seconds for "PlatformIO: Rebuilding IntelliSense Index" to finish for the first time. If this is normal then we can close this issue. The Project Tasks are unusable in this duration which is annoying.

I see this in the console when the problem appears: `[Embedded Page] Failed check of PIO Core library updates for undefined TypeError: Cannot read property 'options' of undefined` Not sure if it is related.

Reproduction steps:

1. Close VS code if open
2. Delete the local `.platformio` folder
3. Launch VS code
4. Wait for PIO to finish installing and reload VS code
5. Observe slow "PlatformIO: Rebuilding IntelliSense Index" (On my end it is around 1 minute and 30 seconds)

This is what it looks like while it is hanging:
[image]

I'm on Windows 10 and Latest VScode

closed 2021-06-17 by stale[bot]

atsju, 2022-06-07:
This is still relevant and no additional question has been asked. No thank you stale bot

ivankravets, 2022-06-07:
@atsju could you reproduce this issue with the latest PlatformIO Core 6.0.2?

atsju, 2022-06-08:
@ivankravets  Core 6.0.2 yes. 
maybe it's relevant, the project was based on ESP32/ESP8266.

atsju, 2022-06-09:
This tag would be a great starting point for discussion https://github.com/luc-github/ESP3D/tree/v3.0-alpha2
There doesn't seem to be lib_deps. But extra script yes https://github.com/luc-github/ESP3D/blob/v3.0-alpha2/platformIO/extra_script.py

atsju, 2022-06-09:
```
PS C:\GIT\ESP3D> C:\Users\myName\.platformio\penv\Scripts\pio project init --ide vscode
Updating metadata for the vscode IDE...
Project has been successfully updated!
PS C:\GIT\ESP3D> 
```
Command did not hang