platformio/platformio-vscode-ide #2516: Start PIO Home Server

rgot-org, 2021-04-14:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.55.2
  PIO IDE: v2.3.0
  System: Windows_NT, 10.0.19043, x64

  %23 Exception
  ```
  Error: Error: HTTPSConnectionPool(host='api.registry.ns1.platformio.org', port=443): Max retries exceeded with url: /v3/packages%3Fquery=type%3A%22tool%22 name%3A%22contrib-piohome%22 (Caused by ProtocolError('Connection aborted.', FileNotFoundError(2, 'No such file or directory')))

    at cmd (c:\Users\francois\.vscode\extensions\platformio.platformio-ide-2.3.0\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\home.js:185:9)
    at ChildProcess.c (c:\Users\francois\.vscode\extensions\platformio.platformio-ide-2.3.0\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:178:7)
    at ChildProcess.emit (events.js:315:20)
    at ChildProcess.EventEmitter.emit (domain.js:483:12)
    at ChildProcess.cp.emit (c:\Users\francois\.vscode\extensions\platformio.platformio-ide-2.3.0\node_modules\cross-spawn\lib\enoent.js:34:29)
    at maybeClose (internal/child_process.js:1021:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)
  ```

rgot-org, 2021-04-14:
Same thing with PIO IDE: v2.3.2

rgot-org, 2021-04-18:
Solved !
the context variable SSLKEYLOGFILE is defined on my system
[image]
but the path as been deleted then the urllib3 librarie was not able to connect to platformio api 
Solved by restoring the path for SSLKEYLOGFILE key

closed 2021-04-18 by rgot-org

sgirad, 2021-08-16:
Hello 
I have same problem.
Please tell me closely.