platformio/platformio-vscode-ide #2523: PlatformIO IDE installation has been suspended, because PlatformIO IDE Installer is already started in another window.

jobe451, 2021-04-18:
VSCode: 1.55.2
PIO IDE: v2.3.2
System: Ubuntu 20.10

When I try to install platformio in vscode, I get this error:
"PlatformIO IDE installation has been suspended, because PlatformIO IDE Installer is already started in another window."

I tried to get rid of possible former instances with:
- pip uninstall platformio
- apt remove platformio
- remove the folder ~/.vscode/extensions

I did find a similar issue with atom:
https://github.com/platformio/platformio-atom-ide/issues/2411

When I try to install the extension from command line, I get

> ```
code --install-extension platformio.platformio-ide
Installing extensions...
bash: Kann die Prozessgruppe des Terminals nicht setzen (-1).: Unpassender IOCTL (I/O-Control) für das Gerät
bash: Keine Job-Steuerung in dieser Shell.
Installing extension 'platformio.platformio-ide' v2.3.2...
(node:8820) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
Extension 'platformio.platformio-ide' v2.3.2 was successfully installed.
```

Fakiridis, 2021-06-20:
I have exactly the same problem in Win 10 pro x64.
I reinstalled both Visual Code and PlatformIO but the problem remains.
Thank you.

mallemang, 2023-04-21:
I've had exactly the same problem windows 10 pro  x64.  I cannot use platformio from within vscode.   I get:  PlatformIO IDE installation has been suspended, because PlatformIO IDE Installer is already started in another window. #2523
I tried uninstalling vscode, platformio but problem remains.   I can go to cmd and type   pio home and that works.  but it won't work with vscode.   I'll give atom a try.

vincentvdschaft, 2024-05-12:
I had this same problem. For me the cause was that there was this line in my settings.json:
```json
"platformio-ide.useBuiltinPIOCore": false,
```
The solution was opening the user settings json and setting this to true.

ntronix44, 2025-02-08:
Setting "platformio-ide.useBuiltinPIOCore": false, up to true didn't work for me either