platformio/platformio-vscode-ide #2540: Cant get extension to install in VSCODE

theone11, 2021-05-03:
Can't get extension to install in VSCODE.
running latest version 1.55.2.

Tried deleting:
C:\Users\<user>\.vscode\extensions\platformio.platformio-ide-2.3.2
C:\Users\<user>\.platformio

tried several times and get the following error during installation phase after a few minutes.

`Error: WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='pypi.org', port=443): Read timed out. (read timeout=15)")': /simple/platformio/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='pypi.org', port=443): Read timed out. (read timeout=15)")': /simple/platformio/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='pypi.org', port=443): Read timed out. (read timeout...`

[image]

N0YHR, 2021-05-05:
.platformIO will not install.  Latest version 2.3.2
Tried several times.  Rebooted, shutdown the VPN temporarily, uninstalled, and tried reinstalling.

[image]

twelti, 2021-06-02:
Similar probleme here, I get an error that no valid CA has been found.

closed 2021-08-08 by stale[bot]