platformio/platformio-vscode-ide #2554: Intermittent can't build issue, shows "Configure Task" popup list

maxgerhardt, 2021-05-13:
Me and multiple people in the community forum are hitting an issue that pressing the "Build" button, either in the sidebar or at the bottom taskbar, does not trigger a `pio run` build but just a VSCode popup asking to configure tasks.

[image]

When further clicking on "PlatformIO" 

[image]

This occurs in e.g.
* https://community.platformio.org/t/need-hor-compilation-with-platform-io/21400?u=maxgerhardt
* https://community.platformio.org/t/clicking-build-open-a-dropdown-list/21239?u=maxgerhardt
* https://community.platformio.org/t/cannot-build-vscode-says-to-configure-a-task/21313?u=maxgerhardt

So, 4 people total in a very short timespan recently. 

At first the fault was intermittent for me and I thought it was related to
* updating VSCode 
* VSCode auto-updating the C/C++ extension to v.1.4.0-insiders

But then intermittently it started working again with all those latest versions, now I'm hitting the same issue again in a new ESP-IDF project.

I've also tried using the "Refresh Project Tasks" button so that PlatformIO maybe contributes tasks again, but it doesn't change it.

What *just* helped was restarting VSCode though and I can build again... Very intermittent issue :( 

VSCode version

[image]

System info 

```
> pio system info
--------------------------  --------------------------------------------------------------------------
PlatformIO Core             5.2.0a6
Python                      3.8.6-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\Max\.platformio
PlatformIO Core Executable  C:\Users\Max\AppData\Local\Programs\Python\Python38\Scripts\platformio.exe
Python Executable           c:\users\max\appdata\local\programs\python\python38\python.exe
Global Libraries            0
Development Platforms       36
Tools & Toolchains          208
```

Extensions

[image]

maxgerhardt, 2021-05-13:
Possibly relevant: https://community.platformio.org/t/pio-vsc-cant-build-no-tasks-found/10488/56?u=maxgerhardt

maxgerhardt, 2021-05-13:
And we've got another one with same issue: https://community.platformio.org/t/esp8266-non-os-sdk-has-errors-when-building-new-project/21404/5?u=maxgerhardt

maxgerhardt, 2021-05-19:
And another one with the same issue: https://community.platformio.org/t/strange-behavior-with-platformio/21515

ivankravets, 2021-05-19:
Could you reproduce this issue with the latest VSCode 1.56.2? I'm not sure that this issue is linked with PlatformIO :( Maybe, VSCode broke Tasks API :(

maxgerhardt, 2021-05-19:
I continously updated over the last timespan and I'm now at 1.56.2, but even so still experienced the issue today. 

[image]

Again, it was solved by a quick VSCode restart. I don't quite know how to reproduce it, but opening a new project or creating a new project in which some dependencies have yet to be downloaded (like the framework) seems to be triggering it (sometimes?). Maybe I can try and get the logs out of the VSCode developer console the next time it occurs.

ivankravets, 2021-05-19:
> but opening a new project or creating a new project in which some dependencies have yet to be downloaded (like the framework) seems to be triggering it (sometimes?).

Yes, you will not see tasks until the project indexing process is not completed. You should see a progress bar below. This is very easy to reproduce with ESP-IDF project which is crazy complex and slow due IDF and its 1Gb source files.

maxgerhardt, 2021-05-19:
However, at some point the process is done and when the "Refresh project tasks" button is used, the tasks refresh correctly in the "project tasks" panel, but clicking on "Build" will still result in the popup that no task was found. This is something I definitely observed.

closed 2021-07-25 by stale[bot]