platformio/platformio-vscode-ide #2558: Extensions calls patchOSEnviron without project and activateOnlyOnPlatformIOProject enabled

septatrix, 2021-05-15:
Platformio seems to load itself even though it is configured not to. Consequently is the PATH env variable modified and the `PLATFORMIO_{IDE,CALLER}` variables are set. I was able to confirm this using:

```shell
$ /home/septatrix/.platformio/penv/bin:/home/septatrix/.platformio/penv:/home/septatrix/.platformio/python3/bin:/home/septatrix/.cargo/bin:/home/septatrix/.cargo/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
$ env | grep PLATFORM
PLATFORMIO_IDE=2.1.0
PLATFORMIO_CALLER=vscode
```

This is the same issue as #2258 however while the problem disappeared it started happening again after a short while. The issue does not occur when the pio extension is not loaded.

Confusingly the checks seem to be properly implemented:

https://github.com/platformio/platformio-vscode-ide/blob/21ab8c9eca5ece45e519d7c3abe47ca66f9ebd0f/src/main.js#L44-L59

I currently do not have the time to add the required logging to the extension and build it myself but I will try to provide more information if possible. While I do not mind the two pio specific env variables the main problem for me is the modification of the path as this changes how pip etc work...

septatrix, 2021-07-15:
If there is some way to get more debugging information I will contribute that alas I do not know of any way to get the currently

septatrix, 2021-09-14:
Again if someone tells me what information is missing I will definitely add that!

ivankravets, 2021-11-05:
How is it possible? We patch the OS environment in https://github.com/platformio/platformio-vscode-ide/blob/develop/src/main.js#L59

Could you reproduce this issue with the latest VSCode + PlatformIO IDe Extension? We updated it today.

septatrix, 2021-11-06:
No I am not able to reproduce this however this also was not deterministic in the past.

However I experienced a similar issue in the past with normal python venv and thinking about it that might be the same what happened here: Sometimes I open a new VSCode instance from the integrated terminal. I might have mistakenly done this from the PIO Terminal which obviously would have led to the environment being preserved and therefore active in the new vscode instance. If that really is the problem I apologize for creating some confusion with this issue as the mistake may likely have been on my side.

closed 2021-11-06 by septatrix