platformio/platformio-vscode-ide #2559: Expose `--skip-packages` option in GUI

maxgerhardt, 2021-05-17:
The PIO Home -> Inspect screen should have an option that makes it so that `--skip-packages` is passed to `pio check`, otherwise ESP32 and SAMD based projects (and possible others) cannot be code-inspected in the UI because `cppcheck` fails.

See https://github.com/platformio/platformio-core/issues/3951 and https://community.platformio.org/t/code-inspection-not-returning-expected-result/21480.

maxgerhardt, 2021-07-22:
This issue is still valid.

maxgerhardt, 2021-07-29:
._. don't close it, dear stale bot.

grisharav, 2021-08-09:
Agree, still there...

maxgerhardt, 2021-08-09:
@ivankravets, how do I reopen the issue if the bot ignores me? :D

MomePP, 2021-11-07:
I got this problem too. Found that there is undocumented check options `check_skip_packages = true` can be set in `platformio.ini`. This passes `--skip-packages` when inspects code from GUI.

Hope this helps, Cheer !

MomePP, 2021-11-07:
I got this problem too. Found that there is undocumented check options `check_skip_packages = true` can be set in `platformio.ini`. This passes `--skip-packages` when inspects code from GUI (PlatformIO Home).

Hope this helps !