platformio/platformio-vscode-ide #2560: ESP32 error in starting debugger

ganeshkumarsw, 2021-05-18:
@ivankravets, thanks for this wonderful tool but currently I am facing a problem while starting the debug session for esp32. 
I request you to kindly support me to solve this issue.

I have followed all the procedure mentioned under
https://community.platformio.org/t/esp32-pio-unified-debugger/4541/19

still at the time of starting the debugger I get the below error

[image]

### platformio.ini

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html
[env:esp-wrover-kit]
platform = espressif32
build_flags = 
	; -DBOARD_HAS_PSRAM
	; -mfix-esp32-psram-cache-issue
	; -DCORE_DEBUG_LEVEL=5
	; -DLOG_LEVEL=LOG_LEVEL_NONE
	; -DCONFIG_ARDUHAL_LOG_COLORS=1
board = esp32-wrover-kit

framework = arduino
; board_build.partitions = default_16MB.csv
upload_port = COM7
monitor_port = COM7
monitor_speed = 480600
monitor_flags = --raw
upload_speed = 921600

debug_tool = esp-prog
debug_init_break = tbreak setup
```

### OpenOCD run through powershell
[image]

### Zadig 
[image]

ivankravets, 2021-05-18:
Please check that you use the latest VSCode + PlatformIO IDE extension.

You have a problem with the installation.

closed 2021-05-18 by ivankravets

ganeshkumarsw, 2021-05-18:
Below with, I have listed platform io extension, core, lib and vscode versions

### Extension
[image]

[image]

### VS code
[image]

ivankravets, 2021-05-18:
This is the problem
[image]

What are your VSCode's settings? Did you change any settings for the PlatformIO IDE extension?

ganeshkumarsw, 2021-05-18:
I have not changed settings for Platform IO extension. I have even tried to debug Platform IO generated project on eclipse IDE and ended up in the below issue

[image]

### VS Code settings.json
```
{
    "workbench.colorTheme": "Monokai",
    "git.autofetch": true,
    "C_Cpp.updateChannel": "Insiders",
    "arduino.additionalUrls": [
        "https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json",
        "http://arduino.esp8266.com/stable/package_esp8266com_index.json",
        "https://raw.githubusercontent.com/VSChina/azureiotdevkit_tools/master/package_azureboard_index.json"
    ],
    "arduino.path": "C:\\\\Program Files (x86)\\\\Arduino",
    "arduino.commandPath": "arduino_debug.exe",
    "arduino.ignoreBoards": [
        "Adafruit HUZZAH ESP8266",
        "MXCHIP AZ3166"
    ],
    "idf.espIdfPathWin": "c:\\Mine\\ESP\\vs-esp-idf-v4.2\\esp-idf",
    "idf.showOnboardingOnInit": false,
    "cmake.configureOnOpen": true,
    "workbench.iconTheme": "vscode-icons",
    "tabnine.experimentalAutoImports": true,
    "extensions.autoUpdate": false,
    "diffEditor.ignoreTrimWhitespace": false,
    "workbench.welcomePage.hiddenCategories": [
        "Setup"
    ],
    "gitlens.advanced.externalDiffTool": "diff.guitool",
    "idf.pythonBinPathWin": "C:\\Mine\\ESP\\.espressif\\python_env\\idf4.2_py3.8_env\\Scripts\\python.exe",
    "idf.customExtraPaths": "",
"idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"C:/Users/rui/.platformio/packages/tool-openocd-esp32@1.1000.20190708/share/openocd/scripts/\"}",
    "idf.toolsPathWin": "C:\\Mine\\ESP\\.espressif",
    "remote.SSH.remotePlatform": {
        "192.168.0.137": "linux"
    },
    "terminal.integrated.tabs.enabled": true,
    "idf.openOcdConfigs": [

        "interface/ftdi/esp32_devkitj_v1.cfg",
        "board/esp32-wrover.cfg"
    ]
}
```

ivankravets, 2021-05-18:
Could you try to disable Espressif's and Arduino extensions in VSCode and try debugging again?

ganeshkumarsw, 2021-05-18:
We are in good sync, Just now I have tried this combination and it worked except I have to reset esp32 all the time for the debugger to get hooked with the hardware

ivankravets, 2021-05-18:
Which extension did you disable?

ganeshkumarsw, 2021-05-18:
I have disabled both Espressif's and Arduino extensions