platformio/platformio-vscode-ide #2570: Copy to clipboard from Library Example Code Doesn't work on Mac VS Code

tgauda, 2021-05-28:
If you attempt to copy any code from a library by either using right click context menu (doesn't pop up) or keyboard shortcut nothing happens.

ivankravets, 2021-06-15:
Sorry for the issue. It was previously reported in https://github.com/platformio/platformio-vscode-ide/issues/606 and we provided a workaround for it.

It seems that a recent update to VSCode has broken WebView API again. See a similar issue https://github.com/microsoft/vscode/issues/65452

@tamuratak do you have any ideas with a new workaround? This (https://github.com/tamuratak/LaTeX-Workshop/commit/e204b0848d0f62cb13175e8e804566e5e1b318b7) workaround does not work anymore.

/cc @bpasero, @mjbvz

bpasero, 2021-06-15:
Thanks for the ping, I defer to Matt for this as he owns that component.

ivankravets, 2021-08-05:
@mjbvz do we have a workaround for https://github.com/microsoft/vscode/issues/65452 ?