platformio/platformio-vscode-ide #2582: Debugging Zephyr with J-Link EDU Mini Hangs on "Starting target CPU..."

pearson, 2021-06-08:
When attempting to debug a Zephyr program using a J-Link EDU Mini, the debugger hangs on "Starting target CPU...":
```
PlatformIO: Resume the execution to `debug_init_break = tbreak main`
PlatformIO: More configuration options -> http://bit.ly/pio-debug
Setting breakpoint @ address 0x00000038, Size = 2, BPHandle = 0x0001
Setting breakpoint @ address 0x00000110, Size = 2, BPHandle = 0x0002
Setting breakpoint @ address 0x0000014A, Size = 2, BPHandle = 0x0003
Starting target CPU...
```
<details>
<summary>Click here for the full Debug Console log from the PlatformIO Blinky example</summary>

```
Processing chris_nrf52832 (platform: nordicnrf52; board: chris_nrf52832; framework: zephyr)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/chris_nrf52832.html
PLATFORM: Nordic nRF52 (8.0.0) > Nordic (Chris) 52832
HARDWARE: NRF52832 64MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) On-board (cmsis-dap, jlink) External (blackmagic, stlink)
PACKAGES:
 - framework-zephyr 2.20500.210226 (2.5.0)
 - framework-zephyr-canopennode 0.0.0-alpha+sha.468d350028
 - framework-zephyr-civetweb 0.0.0-alpha+sha.e6903b80c0
 - framework-zephyr-cmsis 0.0.0-alpha+sha.c3bd2094f9
 - framework-zephyr-fatfs 0.0.0-alpha+sha.1d1fcc725a
 - framework-zephyr-hal-nordic 0.0.0-alpha+sha.f0d54d8449
 - framework-zephyr-hal-st 0.0.0-alpha+sha.b52fdbf4b6
 - framework-zephyr-libmetal 0.0.0-alpha+sha.9d4ee2c3cf
 - framework-zephyr-littlefs 0.0.0-alpha+sha.9e4498d1c7
 - framework-zephyr-loramac-node 0.0.0-alpha+sha.3f545d76a2
 - framework-zephyr-lvgl 0.0.0-alpha+sha.31acbaa36e
 - framework-zephyr-mbedtls 0.0.0-alpha+sha.24d84ecff1
 - framework-zephyr-mcuboot 0.0.0-alpha+sha.3fc59410b6
 - framework-zephyr-mcumgr 0.0.0-alpha+sha.43845e883f
 - framework-zephyr-mipi-sys-t 0.0.0-alpha+sha.75e671550a
 - framework-zephyr-open-amp 0.0.0-alpha+sha.de1b85a130
 - framework-zephyr-openthread 0.0.0-alpha+sha.1d668284a0
 - framework-zephyr-segger 0.0.0-alpha+sha.38c79a447e
 - framework-zephyr-sof 0.0.0-alpha+sha.b5b772dd61
 - framework-zephyr-tfm-mcuboot 1.7.0-rc1
 - framework-zephyr-tinycbor 0.0.0-alpha+sha.40daca97b4
 - framework-zephyr-tinycrypt 0.0.0-alpha+sha.3e9a49d267
 - framework-zephyr-trusted-firmware-m 0.0.0-alpha+sha.96340fb6c0
 - tool-cmake 3.16.4
 - tool-dtc 1.4.7
 - tool-ninja 1.9.0
 - tool-sreccat 1.164.0 (1.64)
 - toolchain-gccarmnoneeabi 1.80201.190214 (8.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in debug mode
Checking size .pio\build\chris_nrf52832\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   8.3% (used 5422 bytes from 65536 bytes)
Flash: [          ]   4.0% (used 20748 bytes from 524288 bytes)
========================= [SUCCESS] Took 1.23 seconds =========================
SEGGER J-Link GDB Server V7.20 Command Line Version

JLinkARM.dll V7.20 (DLL compiled Apr 28 2021 17:34:08)

Command line: -singlerun -if SWD -select USB -device nRF52832_xxAA -port 2331 -speed 3000
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     2331
SWO raw output listening port: 2332
Terminal I/O port:             2333
Accept remote connection:      localhost only
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               on
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 nRF52832_xxAA
Target interface:              SWD
Target interface speed:        3000kHz
Target endian:                 little

Connecting to J-Link...
J-Link is connected.
Firmware: J-Link EDU Mini V1 compiled Feb 18 2021 11:25:23
Hardware: V1.00
S/N: 801029172
Feature(s): FlashBP, GDB
Checking target voltage...
Target voltage: 2.96 V
Listening on TCP/IP port 2331
Connecting to target...
Connected to target
Waiting for GDB connection...
Reading symbols from C:\Users\Chris\PlatformIO\Projects\nRF_Zephyr_Test1\.pio\build\chris_nrf52832\firmware.elf...
undefinedC:\Users\Chris\.platformio\packages\toolchain-gccarmnoneeabi@1.80201.190214\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
PlatformIO Unified Debugger -> http://bit.ly/pio-debug
PlatformIO: debug_tool = jlink
PlatformIO: Initializing remote target...
arch_cpu_idle () at C:\Users\Chris\.platformio\packages\framework-zephyr\arch\arm\core\aarch32\cpu_idle.S:107
107		cpsie	i
Select auto target interface speed (2000 kHz)
Connected to 127.0.0.1
Reading all registers
Read 4 bytes @ address 0x00000F3E (Data = 0xF3BFB662)
Read 2 bytes @ address 0x00000F3E (Data = 0xB662)
Reading 64 bytes @ address 0x00000F00
Received monitor command: clrbp
Received monitor command: speed auto
Select auto target interface speed (2000 kHz)
Received monitor command: reset
Resetting target
Received monitor command: halt
Halting target CPU...
...Target halted (PC = 0x00000950)
Downloading 220 bytes @ address 0x00000000
Downloading 16020 bytes @ address 0x000000DC
Downloading 4044 bytes @ address 0x00003F70
Downloading 8 bytes @ address 0x00004F3C
Downloading 72 bytes @ address 0x00004F44
Downloading 312 bytes @ address 0x00004F8C
Downloading 364 bytes @ address 0x000050D0
Downloading 44 bytes @ address 0x0000523C
Downloading 68 bytes @ address 0x00005268
Resetting target
Loading section rom_start, size 0xdc lma 0x0
Loading section text, size 0x4e60 lma 0xdc
Loading section .ARM.exidx, size 0x8 lma 0x4f3c
Loading section initlevel, size 0x48 lma 0x4f44
Loading section sw_isr_table, size 0x138 lma 0x4f8c
Loading section rodata, size 0x16c lma 0x50d0
Loading section datas, size 0x2c lma 0x523c
Loading section devices, size 0x44 lma 0x5268
Start address 0x950, load size 21152
Transfer rate: 20656 KB/sec, 2350 bytes/write.
Temporary breakpoint 1 at 0x38 (2 locations)
PlatformIO: Initialization completed
Writing register (PC = 0x     950)
Read 4 bytes @ address 0x00000950 (Data = 0xF927F004)
Read 2 bytes @ address 0x00000950 (Data = 0xF004)
Read 2 bytes @ address 0x00000952 (Data = 0xF927)
Reading 64 bytes @ address 0x00000000
Read 4 bytes @ address 0x0000037C (Data = 0xFF12F7FF)
Read 2 bytes @ address 0x00000038 (Data = 0x0D7D)
Read 2 bytes @ address 0x00000110 (Data = 0xB538)
Reading 64 bytes @ address 0x00000100
Read 4 bytes @ address 0x0000017C (Data = 0x000050E8)
Read 2 bytes @ address 0x0000014A (Data = 0xF084)
PlatformIO: Resume the execution to `debug_init_break = tbreak main`
PlatformIO: More configuration options -> http://bit.ly/pio-debug
Setting breakpoint @ address 0x00000038, Size = 2, BPHandle = 0x0001
Setting breakpoint @ address 0x00000110, Size = 2, BPHandle = 0x0002
Setting breakpoint @ address 0x0000014A, Size = 2, BPHandle = 0x0003
Starting target CPU...
```
</details>

My configuration is Windows 10, PlatformIO Core version 5.1.1, and a generic Nordic nRF52832 board. However, [this problem has also been reported](https://github.com/platformio/platform-atmelsam/issues/104) on a system with Ubuntu 19.10, PlatformIO version 4.3.1, and an Adafruit Metro M0 Express (ATSAMD21G18 ) board. 

The Arduino and Mbed sample PlatformIO programs for the nRF52832 will debug properly on my setup, and only Zephyr shows this problem. As well, Zephyr programs can be built and debugged properly with the SEGGER Embedded Studio. It looks like the combination of PlatformIO, Zephyr, and the J-Link EDU Mini is a problem.

Does anyone have suggestions for what might be wrong here?

Thanks!

pearson, 2021-06-08:
This appears to be related to a pop-up error message that appears when starting the debugger:

`'CONFIG_PLATFORM_SPECIFIC_INIT' has unknown type; cast it to its declared type (from data-evaluate-expression CONFIG_PLATFORM_SPECIFIC_INIT)`

When the debugger itself starts and hangs at "Starting target CPU...", VS Code opens `reset.S` at these lines:
```
#if defined(CONFIG_PLATFORM_SPECIFIC_INIT)
    bl z_platform_init
#endif
```

It certainly seems like there's a connection there. I'm at a loss as to what to do about it, though.

Any suggestions would be appreciated!

pearson, 2021-06-08:
This may be related to an error message that has popped up once or twice, but not very regularly:
`'CONFIG_PLATFORM_SPECIFIC_INIT' has unknown type; cast it to its declared type (from data-evaluate-expression CONFIG_PLATFORM_SPECIFIC_INIT)`

When the debugger is hanging on "Starting target CPU...", VS Code opens `reset.S` at these lines:
```
#if defined(CONFIG_PLATFORM_SPECIFIC_INIT)
    bl z_platform_init
#endif
```

There seems like there is a connection there, but I'm at a loss for how to fix it.

Any thoughts on the matter would be appreciated!

pearson, 2021-06-08:
This problem is directly related to the `debug_build_flags`. If `-Og` is set (this is the PlatformIO default), the debugger hangs on "Starting the CPU...". However, the debugger works properly if that flag is set to -O0. 

I'm not clear why this is specifically a problem with Zephyr in PlatformIO, but I'm happy to track the problem further if someone with more knowledge of the system is interested.

closed 2021-08-15 by stale[bot]

zeetabit, 2021-12-27:
Have the same issue, solved by 
```
debug_build_flags=-O0 -g
```