platformio/platformio-vscode-ide #2584: Extension causes high cpu load

nielsnl68, 2021-06-09:
- Issue Type: `Performance`
- Extension Name: `platformio-ide`
- Extension Version: `2.3.2`
- OS Version: `Windows_NT x64 10.0.19042`
- VS Code version: `1.56.2`

:warning: Make sure to **attach** this file from your *home*-directory:
:warning:`c:\Users\HP\AppData\Local\Temp\platformio.platformio-ide-unresponsive.cpuprofile.txt`

Find more details here: https://github.com/microsoft/vscode/wiki/Explain-extension-causes-high-cpu-load

hrydwidar, 2021-06-27:
[platformio.platformio-ide-unresponsive.cpuprofile.txt](https://github.com/platformio/platformio-vscode-ide/files/6721036/platformio.platformio-ide-unresponsive.cpuprofile.txt)

Gaweringo, 2021-08-24:
[platformio.platformio-ide-unresponsive.cpuprofile.txt](https://github.com/platformio/platformio-vscode-ide/files/7035874/platformio.platformio-ide-unresponsive.cpuprofile.txt)

dzarda, 2021-10-08:
Happens when opening a folder with a ESP-IDF project

hrydwidar, 2021-10-09:
thank you @dzarda

dzarda, 2021-11-10:
[platformio.platformio-ide-unresponsive.cpuprofile.txt](https://github.com/platformio/platformio-vscode-ide/files/7515858/platformio.platformio-ide-unresponsive.cpuprofile.txt)

Seems to be triggerred even when an new Code window is opened from within the ESP-IDF enabled environment. Please note that no files/workspace need to be opened in that case. Causes the SSH FS extension to not work (all extensions I suppose)

dzarda, 2021-11-10:
Opening a new fresh Code window (without the IDF env) restores SSH FS operation

dreamer2q, 2021-12-27:
[image]

The terrible CPU usage and load balance even if I have closed the vscode for a long time.

dreamer2q, 2021-12-27:
This caused my computer been in a high temperature in a long time...

CharlesPavlov, 2022-08-08:
[platformio.platformio-ide-unresponsive.cpuprofile.txt](https://github.com/platformio/platformio-vscode-ide/files/9281219/platformio.platformio-ide-unresponsive.cpuprofile.txt)