platformio/platformio-vscode-ide #2586: Peripherial view does not show 8bit addressBlocks

EvilMav, 2021-06-10:
When trying to add AVR peripherial view support  to PIO debugger by providing SVD files converted from Atmel's atdf. PIO seems to not show anything in the peripherial view if the adressBlock is less then 4 bytes.

<details>
<summary>Consider this SVD</summary>

```xml
<?xml version="1.0" encoding="UTF-8"?>
<device>
  <vendor>Atmel</vendor>
  <name>ATmega1284P</name>
  <addressUnitBits>8</addressUnitBits>
  <size>8</size>
  <access>read-write</access>
  <resetValue>0</resetValue>
  <resetMask>0xff</resetMask>
  <peripherals>
    <peripheral>
      <name>AC</name>
      <description>Analog Comparator</description>
      <baseAddress>0x50</baseAddress>
      <addressBlock>
        <offset>0x0</offset>
        <size>0x1</size>
        <usage>registers</usage>
      </addressBlock>
      <registers>
        <register>
          <name>ACSR</name>
          <description>Analog Comparator Control And Status Register</description>
          <addressOffset>0x0</addressOffset>
          <access>read-write</access>
          <fields>
            <field>
              <name>ACIS</name>
              <description>Analog Comparator Interrupt Mode Select bits</description>
              <bitRange>[1:0]</bitRange>
              <access>read-write</access>
              <writeConstraint>
                <useEnumeratedValues>true</useEnumeratedValues>
              </writeConstraint>
              <enumeratedValues>
                <enumeratedValue>
                  <name>VAL_0x00</name>
                  <description>Interrupt on Toggle</description>
                  <value>0</value>
                </enumeratedValue>
                <enumeratedValue>
                  <name>VAL_0x01</name>
                  <description>Reserved</description>
                  <value>1</value>
                </enumeratedValue>
                <enumeratedValue>
                  <name>VAL_0x02</name>
                  <description>Interrupt on Falling Edge</description>
                  <value>2</value>
                </enumeratedValue>
                <enumeratedValue>
                  <name>VAL_0x03</name>
                  <description>Interrupt on Rising Edge</description>
                  <value>3</value>
                </enumeratedValue>
              </enumeratedValues>
            </field>
            <field>
              <name>ACIC</name>
              <description>Analog Comparator Input Capture Enable</description>
              <bitRange>[2:2]</bitRange>
              <access>read-write</access>
            </field>
            <field>
              <name>ACIE</name>
              <description>Analog Comparator Interrupt Enable</description>
              <bitRange>[3:3]</bitRange>
              <access>read-write</access>
            </field>
            <field>
              <name>ACI</name>
              <description>Analog Comparator Interrupt Flag</description>
              <bitRange>[4:4]</bitRange>
              <access>read-write</access>
            </field>
            <field>
              <name>ACO</name>
              <description>Analog Compare Output</description>
              <bitRange>[5:5]</bitRange>
              <access>read-write</access>
            </field>
            <field>
              <name>ACBG</name>
              <description>Analog Comparator Bandgap Select</description>
              <bitRange>[6:6]</bitRange>
              <access>read-write</access>
            </field>
            <field>
              <name>ACD</name>
              <description>Analog Comparator Disable</description>
              <bitRange>[7:7]</bitRange>
              <access>read-write</access>
            </field>
          </fields>
        </register>
      </registers>
    </peripheral>
  </peripherals>
</device>

```
</details>

Nothing gets shown in the view, but the register appears if `addressBlock.size` is set to 4 or more.

I'd be happy to debug, but although according to [Community](https://community.platformio.org/t/platformio-plus-goes-open-source-improving-embedded-development-community-worldwide/8240) PIO Plus is supposed to be open source, I can not find the source of [platformio-vscode-debug](https://www.npmjs.com/package/platformio-vscode-debug). Where is it published?

PS In case of a missing `addressBlock`, [PIO fails](https://github.com/platformio/platformio-vscode-ide/issues/2368) with an extremely unhelpful "Cannot read property '0' of undefined"

maxgerhardt, 2021-06-10:
At https://github.com/Rahix/atdf2svd/issues/25#issuecomment-858741454 I've posted a reference with 8-bit regsiters which works nicely per https://github.com/platformio/platform-ststm8/pull/38, see screenshots.

>I'd be happy to debug, but although according to Community PIO Plus is supposed to be open source, I can not find the source of platformio-vscode-debug. Where is it published? 

+1 on this. I thought I opened an issue about that long ago but I can't find it anymore. I'll open a separate issue for that.

maxgerhardt, 2021-08-09:
~ still valid ~

maxgerhardt, 2021-10-09:
~still valid ~