platformio/platformio-vscode-ide #2587: Open-source the VSCode debugger component code

maxgerhardt, 2021-06-10:
As noted in https://github.com/platformio/platformio-vscode-ide/issues/2586 and , the source code for the component responsible for the debugger integration between the PlatformIO core and VSCode, https://www.npmjs.com/package/platformio-vscode-debug, does not seem to be available in its source form, at least that I know of.

The closest I can get is the tarball at https://registry.npmjs.org/platformio-vscode-debug/-/platformio-vscode-debug-1.3.1.tgz which contains the minified source code, quite unreadable.

```javascript
!function(e,t){"object"==typeof exports&&"object"==typeof module?module.exports=t(require("copy-paste"),require("vscode"),require("vscode-debugadapter"),require("xml2js")):"function"==typeof define&&define.amd?define("platformio-vscode-debug",["copy-paste","vscode","vscode-debugadapter","xml2js"],t):"object"==typeof exports?exports["platformio-vscode-debug"]=t(require("copy-paste"),require("vscode"),require("vscode-debugadapter"),require("xml2js")):e["platformio-vscode-debug"]=t
```

This would make it much easier for the community to track down issues with this code, as e.g. linked with the SVD parser.

EvilMav, 2021-06-14:
@ivankravets @valeros sorry to ping you directly, but these items will be moving nowhere without your personal assistance. **The [forum discussion](https://community.platformio.org/t/platformio-plus-goes-open-source-improving-embedded-development-community-worldwide/8240/18) ends in "Please contact us at contact@platformio.org", but I think this is something that should be properly communicated to the community**. After all, it was claimed open-source and my decision to use it in classes is was influenced by this claim. 

Could you please let us know if we are wasting time trying to improve the platform support withing this component.

maxgerhardt, 2021-08-13:
push

zxuanyu, 2022-08-11:
@maxgerhardt excuse me， platformio-vscode-debug open source? the code is unreadable

maxgerhardt, 2022-08-11:
As far as I know it's minified code, not the original source.

sfilata, 2022-10-27:
Hi, how is it going? @ivankravets