platformio/platformio-vscode-ide #2623: PIO_FRAMEWORK_ARDUINO_MMU_CACHE16_IRAM48_SECHEAP_SHARED unreliable if used in a build

TD-er, 2021-07-02:
I have been playing around with the new esp8266/Arduino 3.0.0 and its new option to create a second heap in 16k IRAM

Using `build_flags = -DMMU_IRAM_SIZE=0xC000 -DMMU_ICACHE_SIZE=0x4000 -DMMU_IRAM_HEAP` is very unreliable with regards to whether it is compiled as such.
Using the define `PIO_FRAMEWORK_ARDUINO_MMU_CACHE16_IRAM48_SECHEAP_SHARED` is already a bit more clear if it is being used or not.

But still it is rather hit or miss if the feature is active or not.

I now use this as part of my PIO configuration:
```
build_flags               = ${esp82xx_3_0_x.build_flags}
                            -DPIO_FRAMEWORK_ARDUINO_MMU_CACHE16_IRAM48_SECHEAP_SHARED 
                            -DUSE_SECOND_HEAP
                            -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK3
                            -Wno-deprecated-declarations
```
The define `USE_SECOND_HEAP` is used in my code to wrap the calls to switch heap, like this:
```c++
    case LabelType::FREE_MEM:               return String(ESP.getFreeHeap());
#ifdef USE_SECOND_HEAP
    case LabelType::FREE_HEAP_IRAM:
      {
        HeapSelectIram ephemeral;
        return String(ESP.getFreeHeap());
      }
#endif
```

It is also used for an enum value and some code to actually get the free heap for DRAM or IRAM.
But with the same code, only a recompile, I do get it to actually show the different free heap values and in a next build it just shows the same value for both given I call above code for both enums.

Just as an illustration a tab I had still open from this morning and a tak of a more recent build running on the same node:
[image]

Earlier build:

<html>
<body>

Free RAM: | 10872 (8688 - sendContentBlocking)
-- | --
Free Heap IRAM: | 17552

</body>
</html>

Later build:
<html>
<body>

Free RAM: | 12288 (10104 - sendContentBlocking)
-- | --
Free Heap IRAM: | 12288

</body>
</html>

novirium, 2022-04-12:
I'm hitting a similar issue, though I've managed to get some more details - in my case, the environment passed when running the "Upload" or "Upload and Monitor" task is not updated _if a monitor is still running_.

Say I have two environments set up in my `platformio.ini` file, and am happily using the task list in VSCode to build and upload both of them. As soon as I have a monitor running (for instance, by running the "Upload and Monitor" task) for one of them, the environment gets frozen for any future "upload" or "upload and monitor" tasks". Running either task will run it for the environment the monitor was running for.

Stopping the monitor allows things to function correctly.

novirium, 2022-04-13:
Further investigation has revealed this is due to the "autoCloseSerialMonitor" feature. When the monitor is resumed, this is done by restoring the task that opened the monitor. If that task was itself an "Upload and Monitor" task, then restoring it will clobber the task you just tried to run - you can actually see this in list of task terminals (if you tried to do an "Upload" it will have failed, and the old Upload and Monitor will have restarted - whether it's the right environment or not). 

Worse, [here](https://github.com/platformio/platformio-vscode-ide/blob/69bfc19bae633213c6d5455b47cb22efb2dcfed4/src/project/tasks.js#L152) the new "Upload and Monitor" task is completely ignored in favour of restoring the old one - even if args like the environment are different.

Fundamentally, I'm not sure it's ever really okay to be trying to restore an "Upload and Monitor" task - as it'll try and re-upload via that, rather than the "Upload" or "Upload and Monitor" task you've just selected...

For now, the main workaround is to just close any active monitor before uploading if you are using multiple environments. The risk here is that people will get stung expecting to have uploaded one environment, when it's actually re-uploaded the old one.

As far as fixing this, I'm also not really sure what the desired behavior here would actually be. Unless there is a way of breaking out the "monitor" task such that it is always independent of which combination task called it? Then it would be able to be restored without the side effects of re-running the entire task that first needed the monitor (like Upload and Monitor).

TD-er, 2022-04-13:
> For now, the main workaround is to just close any active monitor before uploading if you are using multiple environments. The risk here is that people will get stung expecting to have uploaded one environment, when it's actually re-uploaded the old one.

That I can confirm from experience as it happens to me all the time.

reivaxy, 2022-05-11:
This is happening to me all the time as well.
I clicked 3 times in a row on the Upload and Monitor in one task. the first two builds were done on the taks I built before that, and the third on the selected task.
Didn't close terminal or monitor between the 3 clicks.

But it happens almost each time I switch from one task to another.

ivankravets, 2023-02-07:
@TD-er, could you reproduce this issue with the latest PlatformIO IDE 3.0?

TD-er, 2023-02-07:
I will check...

TD-er, 2023-02-07:
Ehhh where do I check the IDE version?
[image]
If you mention this version, then yes, I can reproduce it as I got bitten by it a few times again the last few days.

TD-er, 2023-02-07:
Yep, latest version is installed as far as I can see, it still builds the wrong one if there's still an "Upload and Monitor" session is open.
[image]
First built a "collection A" env, then selected one from my "custom" envs and it started building the "collection A" again.

TD-er, 2023-02-07:
Yep, tested again and still reproducible.

Jason2866, 2023-02-19:
We run in this issue with Tasmota now too, after the custom menu bar option is used (deactivated as quick fix).

nicklasb, 2023-02-21:
@ivankravets Chiming in with me encountering this as well, to add that it is quite annoying; one of the main points for PlatformIO is running the same stuff on different boards and platforms in the same integrated environment, so I would agree with @Jason2866 that it is a bug.

nicklasb, 2023-02-24:
> @Jason2866, @nicklasb , could you try [PlatformIO IDE 3.0.1-beta.1 for VSCode](https://www.dropbox.com/s/c9gf04iyjz5i695/platformio-ide-3.0.1-beta.1.vsix?dl=1) build?
> 
> 1. Download the VSIX file above
> 2. Follow to "Extensions" activity bar, `...`, Install from VSIX...
> 
> Restart VSCode. Use the next snippet for custom toolbar:
> 
> ```json
>     "platformio-ide.toolbar": [
>         {
>             "text": "$(debug-alt)",
>             "tooltip": "PlatformIO: Upload & Monitor",
>             "commands": "platformio-ide.uploadAndMonitor"
>         }
>     ]
> ```
> 
> Does it work?

Hi, I installed, however I don't know PlatformIO internals, so I don't know where to put the snippet?
(just installing didn't help)

Kind regards, Nicklas

Jason2866, 2023-02-25:
@ivankravets "Upload and Monitor" does work now. But the issue with not using the env which is choosen from the bottom menu is still there.

[image]

ivankravets, 2023-02-25:
> But the issue with not using the env which is choosen from the bottom menu is still there.

Could you provide more details on how to reproduce it? Or, maybe, simple project?

Jason2866, 2023-02-25:
Defining a few [env] in platformio.ini. Set a one as default. Override this selection by selecting a different one in bottom menu bar (in the screenshot above it is "tasmota32-lvgl") and the default [env]  will be used.

This way  is working fine when removing the custom toolbar.

TD-er, 2023-02-25:
Isn't that the same as the original report?

- upload & monitor envA
- Do not close this console
- upload & monitor envB => envA is being built and uploaded

ivankravets, 2023-02-25:
I've reproduced this bug. Thanks. We will solve this issue.

TD-er, 2023-02-25:
Yay!
Like.... I've been bitten by this bug several times a day.
Quite frustrating to be confronted with your own stupidness for such a long time, so would be great to see this fixed :)

closed 2023-02-27 by ivankravets

ivankravets, 2023-02-27:
Please re-test the latest build:

1. Download [PlatformIO IDE 3.0.1-beta.2.vsix](https://www.dropbox.com/s/qxzysarz634mq75/platformio-ide-3.0.0-beta.2.vsix?dl=1)
2. Follow to VSCode "Extensions" activity bar, `...`, Install from VSIX...
3. Restart VSCode.

Does it work now?

Jason2866, 2023-02-27:
@ivankravets Does not fix. Issue is still the same.
[image]

Jason2866, 2023-02-27:
@ivankravets  Is there a possibility to use added targets from a platform [like platform-espressif32](https://github.com/platformio/platform-espressif32/blob/develop/builder/main.py#L509-L526) in a custom toolbar?

nicklasb, 2023-02-27:
@ivankravets  Hi! 
Tested this and found some problems:
**My boards:**
1. It does create a new terminal and it uses the right target, which is good. 
2. However, if you try with yet another target, the second terminal closes and the new takes its place. 
3. If I click the first one, both are closed and only the last one runs.

2-3 are problematic, as I want to monitor several targets on different boards. 

**Native platform:**
4. The `native`platform now seems to have an "Upload and monitor" choice, which is nice as it runs it. 
5. However, that platform is just an operating system process, so there is nothing to monitor, so the monitor is for some reason connected to what seems to be the last thing I monitored before, which is one of my boards.

ivankravets, 2023-02-27:
@Jason2866, yes, just use the name of the task which you see in the VSCode Menu: Terminal > Run Task > PlatformIO

```json
        "text": "$(check)",
        "tooltip": "PlatformIO: Build",
        "commands": [
          {
            "id": "workbench.action.tasks.runTask",
            "args": "PlatformIO: Upload Filesystem Image (tasmota32-lvgl)"
          }
        ]
      
```

Jason2866, 2023-02-27:
@ivankravets Thx for the answer, is there a variable holding the current env which can be used instead of hard coded value here?

ivankravets, 2023-02-27:
@Jason2866, good idea. That we can remove "custom commands", such as `platformio-ide.uploadAndMonitor`. 

Could I ask yo to file a feature request? We will implement it in `3.0.1` release. It will allow users to use native VSCode's `workbench.action.tasks.runTask` and PlatformIO commands.

```
PlatformIO: Upload Filesystem Image (${currentEnvironment})
```

nicklasb, 2023-02-28:
@ivankravets Nice! 
Works much better now, if I click on "Upload and monitor" on the different boards, it will correctly re-use the correct one.

The only kink is that "Native" still starts a connection to one of my boards (the first one in platformio.ini), and that will later likely block the upload of "Upload and monitor" as it hogs that port. 
Not difficult to work around, but annoying to have to think remember.

In its current state,perhaps the Native platforms' "Upload and monitor" should have a "Run"-task instead to separate it properly? Unless there is some upload support for Native to for example som RaspberryPy or something?

mryall-mawson, 2023-03-03:
Thanks for fixing this. I tried 3.0.1-beta.3 in my project with two devices/environments and the build targets now use the correct environment and open or reuse the right terminal windows.

I've just started using PlatformIO this week, so was great to see this fixed. It makes really simple to deploy and monitor on two different devices now - kudos!

ivankravets, 2023-03-07:
> The only kink is that "Native" still starts a connection to one of my boards (the first one in platformio.ini), and that will later likely block the upload of "Upload and monitor" as it hogs that port.
Not difficult to work around, but annoying to have to think remember.

@nicklasb, could you share here the `platformio.ini` contents to reproduce this issue?

nicklasb, 2023-03-07:
> @nicklasb, could you share here the `platformio.ini` contents to reproduce this issue?
Sure! Here is the project (linked to it before, but realized that was an edit).

You *should* be able to run at least Native out of the box, the project is sort of a start-of-point so it should work:
https://github.com/RobustoFramework/Multi-platform-Multi-board

ivankravets, 2023-03-10:
The PlatformIO IDE 3.0.1 (pre-release version of 3.1.0) is published to the registry. 

@nicklasb, it contains the fix for the "native" platform. Please re-test:

1. Navigate in VSCode to the Extensions activity bar
2. Click on the PlatformIO IDE
3. Switch to the PRE-RELEASE version.

Does it work now?

nicklasb, 2023-03-10:
@ivankravets 
Strange, it doesn't seem to be possible to choose the pre-release (3.0.1) anywhere. 
I don't presume that the setting "use Development PIO Core" is relevant as I gather this is not a PIO Core change, and that doesn't affect my choices.
Is there something more I should do?

Jason2866, 2023-03-10:
@nicklasb In VSC extensions switch to Pre-Release Version

nicklasb, 2023-03-10:
Interestingly, I don't see that button. Is that because I run the .vsx?
[image]

Jason2866, 2023-03-10:
@nicklasb "update" to 3.0.0 and after this you can change to prerelease

nicklasb, 2023-03-11:
Ok, now I have managed to switch, but the problem persists, regrettably. 

Note the log below, after `[SUCCESS]` it tries to connect to `/dev/cu.usbmodem142401 | 9600 8-N-1` which causes all the garbage data as that the wrong speed for that device. 
I don't want to carry on, but could you not reproduce using this [project](https://github.com/RobustoFramework/Multi-platform-Multi-board)? It should try to connect to some (hopefully) non-existing usb-port on your host. 

```
Linking .pio/build/native/program
Configuring upload protocol...
Executing .pio/build/native/program
I (6) Test_Tag: Before delay 1001
I (1006) Test_Tag: After delay 1002
================================================================ [SUCCESS] Took 6.57 seconds ================================================================
--- Terminal on /dev/cu.usbmodem142401 | 9600 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
q4꥗a3����B"�rRH
               (�Hp P+� ���0���$�j�<�
```

Jason2866, 2023-03-11:
@nicklasb  Have you tried to add the monitor speed to the port?
```
[env:uno]
; Arduino Uno Rev3
platform = atmelavr
board = uno
framework = arduino
lib_extra_dirs = components

monitor_port = /dev/cu.usbmodem142401
upload_port = /dev/cu.usbmodem142401
```

Or maybe i do not understand your issue at all?

nicklasb, 2023-03-11:
@Jason2866  
I afraid that is not the problem, it should not connect to the port at all.
I am running the native platform, it should not randomly connect to (and lock) recently used USB ports.

Jason2866, 2023-03-11:
Imho it is a platformio user config error to use `upload and monitor` for native.

ivankravets, 2023-03-11:
Sorry, I don't understand the problem at all :(  See this commit what we did https://github.com/platformio/platformio-vscode-ide/commit/da93a68163a29db78b0419559c4dba0b25c8909e

It means, when there is an open serial port monitor and you try to use a "native" dev-platform, we will not close the active monitors.

It seems we should remove that part of the code because the use case is very rare.

nicklasb, 2023-03-12:
@ivankravets @Jason2866 
Actually, I just saw that "Build" runs the code now (it wasn't my impression it did earlier?). 
So "Upload and monitor" could be removed without any consequence, at least for me.

birnam, 2023-03-15:
This fix appears to have created undesireable behavior. As of 3.1 (I've rolled back to 3.0 and verified this is new to this version), clicking on "Upload and Monitor" subsequent times causes a "The task ... is already active. [Terminate Task] [Restart Task]" dialog.
[image]

nicklasb, 2023-03-15:
@birnam 
> This fix appears to have created undesireable behavior. 
I am getting this aswell.

ivankravets, 2023-03-15:
@birnam , could I ask you to open a new issue at https://github.com/platformio/platformio-vscode-ide/issues?

birnam, 2023-03-15:
@ivankravets @nicklasb New issue created here: https://github.com/platformio/platformio-vscode-ide/issues/3656

Adrianotiger, 2025-03-07:
> This fix appears to have created undesireable behavior. As of 3.1 (I've rolled back to 3.0 and verified this is new to this version), clicking on "Upload and Monitor" subsequent times causes a "The task ... is already active. [Terminate Task] [Restart Task]" dialog. [image]
> 
> This appears to be a regression of this bug: [#2266](https://github.com/platformio/platformio-vscode-ide/issues/2266)

With this command: ``workbench.action.terminal.kill``, you can solve it:
https://community.platformio.org/t/close-serial-terminal-when-uploading-in-vs-code/3635/11

Example:
```
{
            "text": "$(send)",
            "tooltip": "PlatformIO: Upload and Monitor",
            "commands": [
                "workbench.action.terminal.kill",
                "platformio-ide.uploadAndMonitor"
            ]
        },
```

commit 9b0ac41b (2023-02-27): Fixed an issue when the "Upload & Monitor" task selects the wrong environment // Resolve #2623