platformio/platformio-vscode-ide #2640: Turning off updates doesn't work

SameonPenner, 2021-07-30:
Issue Type: <b>Bug</b>

I am a guy hardly having internet so when I go out to program a board almost always updates start running which blocks
 uploading to board and compiling. I have triple checked that auto update is unchecked.

[image]
[image]
Extension version: 2.3.1
VS Code version: Code 1.56.2 (054a9295330880ed74ceaedda236253b4f39a335, 2021-05-12T17:13:13.157Z)
OS version: Windows_NT x64 10.0.19043

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i7-6600U CPU @ 2.60GHz (4 x 2808)|
|GPU Status|2d_canvas: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>oop_rasterization: enabled<br>opengl: enabled_on<br>rasterization: enabled<br>skia_renderer: enabled_on<br>video_decode: enabled<br>vulkan: disabled_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|7.82GB (2.70GB free)|
|Process Argv|--crash-reporter-id 77f8c20e-16e3-48a0-9a3c-cd00dba355b1|
|Screen Reader|no|
|VM|0%|
</details>

HammerPiano, 2021-08-14:
In the main file, startup function, the code only checks if the currently opened directory is a valid platfformio project https://github.com/platformio/platformio-vscode-ide/blob/master/src/main.js#L60

I think it is good to also check in the vscode settings it self, if the user has disabled auto updates. Another way is to add it as a settings whether to automatically check for updates on startup.

ivankravets, 2021-09-02:
The settings which you changed relate to VSCode, not to the PlatformIO IDE extension. It seems you want to disable checking updates for libraries and dev-platforms. See https://docs.platformio.org/en/latest/core/userguide/cmd_settings.html#settings

closed 2021-09-02 by ivankravets