platformio/platformio-vscode-ide #2665: Does not work in Github Codespaces

Glavin001, 2021-08-17:
## Problem

When opening PlatformIO Home:

[image]

VSCode extension appears to load from `127.0.0.1` instead of the GitHub Codespaces URL.

## Applicable Code

I think this comes from:
- `iframe src`: https://github.com/platformio/platformio-vscode-ide/blob/69bfc19bae633213c6d5455b47cb22efb2dcfed4/src/home.js#L158-L162
- `getFrontendUrl`: https://github.com/platformio/platformio-node-helpers/blob/b2a974c350bd5cfa2b680df5fdd762dbeef9df07/src/home.js#L46-L58
- `constructServerUrl`: https://github.com/platformio/platformio-node-helpers/blob/b2a974c350bd5cfa2b680df5fdd762dbeef9df07/src/home.js#L33-L43
- `_HTTP_HOST`: https://github.com/platformio/platformio-node-helpers/blob/b2a974c350bd5cfa2b680df5fdd762dbeef9df07/src/home.js#L29

```javascript
let _HTTP_HOST = '127.0.0.1';
```

ivankravets, 2021-09-02:
You can change this value in settings. See screenshot below

[image]

yhur, 2022-10-03:
Hi @ivankravets 

The #4431 may be slightly different from this, but may be resolved by the resolution for this issue.

For the clarification, what I did was 
1. installed the code-server on RPi
2. Downloaded the PIO from marketplace and uploaded to RPi
3. Accessed the code-server with Safari
4. Installed the PIO manually with the VSIX file
5. Restarted the code-server

And as I stated, I can use the PIO/code-server to develop for esp8266 Arduino successfully. That's great and thank you for that. This seems the core part of PIO is working. 

What's missing is the PIO Home pages. I changed the HTTP_HOST to 0.0.0.0 also, but the PIO home page is not coming.

I always appreciate the work of you and your team. Thanks.

neilenns, 2022-11-10:
I just hit this as well, and I expect others will now that Codespaces is generally available in the free tier of GitHub.

mikethebee, 2022-12-31:
> I just hit this as well, and I expect others will now that Codespaces is generally available in the free tier of GitHub.

As GitHub have been prompting me to try Codespaces/VSC and I normally use PIO locally, I have just come across the this issue too. Changing host in setting times out rather than refused when using 127. on a PC within a Chrome Browser. Any suggestions?

konkrog, 2023-03-08:
Also came across this issue trying to use PlatformIO on Github CodeSpaces... any fixes planned for this?

theo-real, 2023-04-12:
Any fixes planned for this?

Bob2345de, 2024-04-23:
Any solutions for this?

bilogic, 2024-06-15:
1. On my code-server's terminal, I see this process running `python -m platformio -c vscode home --port 45334 --host 127.0.0.1 --session-id xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx --shutdown-timeout 3600 --no-open`
2. It is listening only to `127.0.0.1` of the server, but the VSCode on our client is trying to connect to the client's `127.0.0.1` and there-in lies the problem
3. @ivankravets Can the python command be configured from VSCode?

Thanks!

bilogic, 2024-06-15:
> [image]

After changing this to the process's machine's IP address, and restarting the machine, it now runs as `python -m platformio -c vscode home --port 45334 --host 172.1.1.1 --session-id xxxxxxxxxxxxxxxxxxxxxxxxxx --shutdown-timeout 3600 --no-open`

And now, clicking on PlatformIO's `PIO Home` > `Open` gives me a blank page instead of the error.
From my client, I was able to connect to the IP and port and confirm it runs a HTTP server.

bilogic, 2024-06-15:
Here are my conclusions:

1. Executing `pio home` runs a process like this `python /home/dv1/.platformio/penv/bin/pio home` and the port is automatically captured by code-server on the server and forwarded to the client, that is the reason why I was able to load the usual PIO Home page in my client's browser
2. I believe a fix is needed to get `code-server` to also automaticlaly forward the port after PIO starts the `python -m platformio -c vscode home --port 45334 --host 172.1.1.1 --session-id xxxxxxxxxxxxxxxxxxxxxxxxxx --shutdown-timeout 3600 --no-open` process
3. As far as I can tell, PIO currently cannot upload to the client's com ports (this is a show stopper), would need to see how ESPHome does it
4. I'm not sure if codespaces/vscode.dev allow us to start a python process on their servers, so it might never be possible to run PIO there, but uploading compiled firmware to any client connecting to `code-server` via Chrome is likely possible

bilogic, 2024-10-31:
This supports webserial https://open-vsx.org/extension/eclipse-cdt/serial-monitor 
If PIO can also support webserial, then we can use `local` Chrome to develop + compile on `remote` codespaces, then `Upload and monitor` our local MCUs dev boards.

bilogic, 2025-07-03:
@jhordies how do you deal with uploading the firmware from the server to your local COM port?

bilogic, 2025-07-05:
@jhordies ok virtual here usb will work, although I was hoping to just rely on chrome. Thanks!

jhordies, 2025-07-05:
I'm more interested in the console than uploading.

Build remotely, find the binary in your project, right click download,...
Then flash locally

segfly, 2025-07-05:
If webserial support was implemented in the PIO vscode extension, it would allow for both serial console and flashing. It appears the [ESP-IDF extension](https://marketplace.visualstudio.com/items?itemName=espressif.esp-idf-web) managed to get it working - so technically it should be possible for the PIO extension.

The question becomes how do we get this request prioritized?