platformio/platformio-vscode-ide #2667: Start PIO Home Server

TriCri, 2021-08-18:
%23 Description of problem
  ALready tried to re-install VS Code, delete folder 'platformio.platformio-ide-2.3.3' in extensions folder an re-install the platformIO extension.

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.59.0
  PIO IDE: v2.3.3
  System: Windows_NT, 10.0.19043, x64

  %23 Exception
  ```
  Error: Could not start PIO Home server: Error: connect EACCES 127.0.0.1:8010
	at c:\Users\chris\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:5946
	at processTicksAndRejections (internal/process/task_queues.js:93:5)
  ```

longforus, 2021-08-19:
i have the same issue,

```
stack trace: RequestError: connect ECONNREFUSED 127.0.0.1:8027	at ClientRequest.<anonymous> (c:\Users\void\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\got\dist\source\core\index.js:956:111)	at Object.onceWrapper (events.js:422:26)	at ClientRequest.emit (events.js:327:22)	at ClientRequest.EventEmitter.emit (domain.js:467:12)	at ClientRequest.origin.emit (c:\Users\void\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\@szmarczak\http-timer\dist\source\index.js:39:20)	at Socket.socketErrorListener (_http_client.js:469:9)	at Socket.emit (events.js:315:20)	at Socket.EventEmitter.emit (domain.js:467:12)	at emitErrorNT (internal/streams/destroy.js:106:8)	at emitErrorCloseNT (internal/streams/destroy.js:74:3)	at processTicksAndRejections (internal/process/task_queues.js:80:21)	at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1146:16)
S @ workbench.desktop.main.js:sourcemap:62
workbench.desktop.main.js:sourcemap:2038 [[object Object]]connect ECONNREFUSED 127.0.0.1:8027
$onExtensionRuntimeError @ workbench.desktop.main.js:sourcemap:2038
workbench.desktop.main.js:sourcemap:2038 RequestError: connect ECONNREFUSED 127.0.0.1:8027
	at ClientRequest.<anonymous> (c:\Users\void\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\got\dist\source\core\index.js:956:111)
	at Object.onceWrapper (events.js:422:26)
	at ClientRequest.emit (events.js:327:22)
	at ClientRequest.EventEmitter.emit (domain.js:467:12)
	at ClientRequest.origin.emit (c:\Users\void\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\@szmarczak\http-timer\dist\source\index.js:39:20)
	at Socket.socketErrorListener (_http_client.js:469:9)
	at Socket.emit (events.js:315:20)
	at Socket.EventEmitter.emit (domain.js:467:12)
	at emitErrorNT (internal/streams/destroy.js:106:8)
	at emitErrorCloseNT (internal/streams/destroy.js:74:3)
	at processTicksAndRejections (internal/process/task_queues.js:80:21)
	at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1146:16)
$onExtensionRuntimeError @ workbench.desktop.main.js:sourcemap:2038
```

longforus, 2021-08-30:
@henrygab After a few days, he's all right now. 😄

henrygab, 2021-08-30:
I'm still hitting this.  I understand this can be difficult to diagnose remotely.  I'd love hints on how to do so.

From black-box testing:
1. cmd-line invocation of `pio home` loads the web page, which reports `ERR_CONNECTION_REFUSED`
    a. after reboot, `pio home` appears to start the server...
    b. after running once (either cmd-line or failing to open page in VSCode), reports `PlatformIO Home server is already started in another process`
2. However, access to `https://127.0.0.1:8008` (the configured PIO Home port) returns `ECONNREFUSED`
3. Attempts to access `https://127.0.0.1:31524` (random port with nothing) gets the same `ECONNREFUSED`

Based on the above, I can't tell if the PIO Home server is **_actually_** non-responsive, or if something if interfering with the connection, which would be my first step at diagnosis.

### Is there a way to cause the PIO Home server to be periodically logging a "keepalive" type message to a log somewhere, just to verify it's alive and well?

henrygab, 2021-10-30:
It appears that PIO folks don't review questions asking how to get data to enable tracing the root cause of issues.  That makes me sad, while I simultaneously understand the problem of high-volume feedback.

For me, this eventually resolved itself after many weeks.  I have no idea why.  I really don't like not knowing the root cause, because that means when it occurs again, I will again be stuck with no way forward.

PIO folk ... if you have any data that you want collected (or would be helpful to collect) when this occurs, **_PLEASE_** post here.  I'm not the only one this happens to, and it prevents use of PIO entirely when it does happen.  Else, I won't update again next time this goes stale.

ivankravets, 2021-11-05:
Could someone provide us with a remote session to debug this issue? Please email to me@ikravets.com for details.

henrygab, 2021-11-20:
After a time of not occurring, it is happening again.  I've reached out to Ivan Kravets to see how to enable further debugging.

henrygab, 2021-11-22:
Some (potentially) useful information:

I'm on Windows 10.  The TCP port that `PIO Home` was attempting to run on was port `8080`.

However, Windows may have reserved that port for some other program?  (not entirely sure about this part...)

See output of the following command (from a cmd.exe terminal):

`netsh int ipv4 show excludedportrange tcp`

I my case, ports `7981    8080` were listed.  I was able to workaround the issue (at least temporarily) by manually setting the port used by PIO Home to a number that was *not* listed in the ranges from the above command.  In my case, I used port `11011`.

Still seeing if there is anything more I can do to help root-cause more deeply, hopefully with the goal of preventing this silent, non-actionable error from affecting other users....

henrygab, 2021-11-23:
After more digging, it seems highly likely to be related to [issue 3171](https://github.com/docker/for-win/issues/3171#issuecomment-554587817), as reported in the docker repository.

<details><summary>Really short summary of how combination of issues may cause this....</summary>

If 3171 turns out to be the same cause, then this issue is caused by a combination of Hyper-V (specifically Hyper-V Networking Service aka HNS) reserving a LOT of dynamic ports, coupled with a botched Windows update (possibly the April 2020 Windows 10 (aka 2004)?), which was intended to update both the starting TCP and UDP ports for dynamic binding to be 49152 ... but something goes wrong and TCP is left at old setting that allocates dynamic TCP ports starting at 1024.

First, it's perfectly valid for HNS to reserve a bunch of dynamic ports.  The problem is, when they could include values that are used by other programs that expect a fixed port.  Thus, because HNS might reserve a different subset of TCP ports at each and every boot, it will cause random-looking failures.  Debugging is very hard because most utilities won't show that anything is actually using those ports, but attempting to bind to them will still fail.

Wow!  I hope this is the cause!

</details><HR/>

<details><summary>Minimal fix if TCP dynamic ports start at 1024?</summary>

Here's the output from the relevant commands on my machine:
```
C:\Q>netsh int ipv4 show dynamicport tcp

Protocol tcp Dynamic Port Range
---------------------------------
Start Port      : 1024
Number of Ports : 64511

C:\Q\>netsh int ipv4 show dynamicport udp

Protocol udp Dynamic Port Range
---------------------------------
Start Port      : 49152
Number of Ports : 16384
```

As is clear, the starting port # for dynamically allocated TCP ports was set to start at 1024.  It appears that Windows updated the UDP settings to a reasonable range, but failed to update the TCP settings.  Running the following (from an administrator/elevated cmd.exe) updates the starting port number for TCP:

`netsh int ipv4 set dynamic tcp start=49152 num=16384`

Will that fix it?  I'm about to find out (reboot pending).

</details><HR/>

<details><summary>Alternate issue from docker uninstall?</summary>

Apparently, uninstalling Docker ... doesn't cleanly uninstall docker.  If you had installed docker on Windows 10, then uninstalled Docker, and the above TCP port fix doesn't help, then the following may be worth review.  (I've not tried this.)

https://docs.microsoft.com/en-us/virtualization/windowscontainers/manage-docker/configure-docker-daemon#clean-up-docker-data-and-system-components

</details><HR/>

<details><summary>Alternate issue via Hyper-V?</summary>

See https://stackoverflow.com/a/64992483/6471040.

This links to https://dandini.wordpress.com/tag/administered-port-exclusions/, which describes another registry key that (apparently) disables some HNS reservations?

</details><HR/>

Will update this post with results.

henrygab, 2021-11-23:
@ivankravets ... I think I found the root cause.

Can you detect if the dynamic TCP start port is 1024, and if so, provide a useful error message on how a person can fix this themselves?

henrygab, 2021-11-25:
@topilin -- Does the above fix your issue also?

henrygab, 2021-11-30:
@longforus ... if you've not fixed the root cause, please see [above solution](https://github.com/platformio/platformio-vscode-ide/issues/2667#issuecomment-976142717).

ivankravets, 2022-12-26:
Should be fixed in upcoming PlatformIO IDE 3.0

closed 2022-12-26 by ivankravets