platformio/platformio-vscode-ide #2682: Start PIO Home Server

jasondouglasdragoon, 2021-09-09:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.60.0
  PIO IDE: v2.3.3
  System: Linux, 5.4.0-81-generic, x64

  %23 Exception
  ```
  Error: Error: Traceback (most recent call last):
  File "/home/jason/.platformio/penv/lib/python3.6/site-packages/platformio/__main__.py", line 109, in main
    cli()  %23 pylint: disable=no-value-for-parameter
  File "/home/jason/.platformio/penv/lib/python3.6/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/jason/.platformio/penv/lib/python3.6/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/jason/.platformio/penv/lib/python3.6/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/home/jason/.platformio/penv/lib/python3.6/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/jason/.platformio/penv/lib/python3.6/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/jason/.platformio/penv/lib/python3.6/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/jason/.platformio/penv/lib/python3.6/site-packages/platformio/commands/home/command.py", line 98, in cli
    home_url=home_url,
  File "/home/jason/.platformio/penv/lib/python3.6/site-packages/platformio/commands/home/run.py", line 63, in run_server
    contrib_dir = get_core_package_dir("contrib-piohome")
  File "/home/jason/.platformio/penv/lib/python3.6/site-packages/platformio/package/manager/core.py", line 48, in get_core_package_dir
    pkg = pm.get_package(spec)
  File "/home/jason/.platformio/penv/lib/python3.6/site-packages/platformio/package/manager/base.py", line 236, in get_package
    for pkg in self.get_installed():
  File "/home/jason/.platformio/penv/lib/python3.6/site-packages/platformio/package/manager/base.py", line 209, in get_installed
    pkg = PackageItem(pkg_dir)
  File "/home/jason/.platformio/penv/lib/python3.6/site-packages/platformio/package/meta.py", line 405, in __init__
    self.metadata = self.load_meta()
  File "/home/jason/.platformio/penv/lib/python3.6/site-packages/platformio/package/meta.py", line 440, in load_meta
    return PackageMetaData.load(manifest_path)
  File "/home/jason/.platformio/penv/lib/python3.6/site-packages/platformio/package/meta.py", line 391, in load
    data = json.load(fp)
  File "/usr/lib/python3.6/json/__init__.py", line 299, in load
    parse_constant=parse_constant, object_pairs_hook=object_pairs_hook, **kw)
  File "/usr/lib/python3.6/json/__init__.py", line 354, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.6/json/decoder.py", line 339, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.6/json/decoder.py", line 357, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at /home/jason/.vscode/extensions/platformio.platformio-ide-2.3.3/node_modules/platformio-node-helpers/dist/index.js:1:5860
	at ChildProcess.a (/home/jason/.vscode/extensions/platformio.platformio-ide-2.3.3/node_modules/platformio-node-helpers/dist/index.js:1:771284)
	at ChildProcess.emit (events.js:315:20)
	at ChildProcess.EventEmitter.emit (domain.js:467:12)
	at maybeClose (internal/child_process.js:1048:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
  ```

ivankravets, 2021-11-05:
Could you try to remove `~/.platformio/packages/contrib-piohome` package and restart VSCode. Does it work now?

P.S: Please update VSCode + PlatformIO IDE extension to the latest versions.

closed 2022-01-12 by stale[bot]

dalscho13, 2023-06-24:
I realize this issue is closed...but I am having the same exact issue now. 

I did use the terminal to remove 'contrib-piohome' and that didn't make a difference. Any further ideas?

I have the updated VSCode and PlatformIO extensions.