platformio/platformio-vscode-ide #2683: Bundled Python for the Terminal

yhur, 2021-09-14:
I have recently noticed my Windows and my MacOS showed some discrepancy of picking up a wrong Python in the PIO terminal. 

My conclusion is that the Windows installation puts the ~/.platformio/penv/Scripts at the beginning of the %PATH% environment variable so it picked up the right one while my MacOS installation puts the ~/.platformio/penv/bin later than the system wide python folder in the $PATH variable in the PIO terminal hence picking up the system wide python which I think is wrong in this context. 

So the suggestion is to implement some logic that does this.

1. when click to open a terminal in the PIO, check the python 
<pre>
which python | grep \.platformio\/penv > /dev/null
</pre>
2. if the picked up python is not the PIO bundled one, then source the ~/.platformio/penv 
<pre>
source ~/.platformio/penv/bin/activate
</pre>
3. and open the terminal with this environment

yhur, 2021-09-16:
After playing around for some more time, I come to 100% that that it’s an error. I noticed whenever a PIO terminal gets open, there is some algorithm running already which sets the PATH variable.

While that logic should put the ~/.platformio/python3 ahead of any system wide python directory in the PATH, somehow it's failing.

ivankravets, 2021-11-05:
I experience the same issue on macOS. I'll think about how to fix it.