platformio/platformio-vscode-ide #2684: Start PIO Home Server

aleh1129, 2021-09-15:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.60.1
  PIO IDE: v2.3.3
  System: Windows_NT, 6.1.7601, x64

  %23 Exception
  ```
  Error: Could not start PIO Home server: Error: getaddrinfo ENOTFOUND 127.0.0.1:810
	at c:\Users\User\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:5946
	at processTicksAndRejections (internal/process/task_queues.js:93:5)
  ```

dbagnara, 2021-09-15:
I am getting a similar problem this morning. Windows forced a reboot of the PC overnight and pio hasn't worked since.
Even from the command prompt, pio -h returns Error: No such command "access"

 VSCode: 1.60.0
  PIO IDE: v2.3.3
  System: Windows_NT, 10.0.19042, x64

  %23 Exception
  ```
  Error: Usage: __main__.py [OPTIONS] COMMAND [ARGS]...
Try '__main__.py -h' for help.

Error: No such command "home"

	at c:\Users\David xxxxxx\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:5860
	at ChildProcess.a (c:\Users\David xxxxx\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771284)
	at ChildProcess.emit (events.js:315:20)
	at ChildProcess.EventEmitter.emit (domain.js:467:12)
	at ChildProcess.cp.emit (c:\Users\David xxxxxx\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1048:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
  ```

ivankravets, 2021-11-05:
1. Remove global PlatformIO Core, see https://docs.platformio.org/en/latest/faq.html#multiple-platformio-cores-in-a-system
2. Please update VSCode to the latest 1.62.0 or above version, and install the latest (2.4.0 or above) PlatformIO IDE extension. See https://marketplace.visualstudio.com/items/platformio.platformio-ide/changelog

Does it work now?

closed 2021-11-05 by ivankravets