platformio/platformio-vscode-ide #2691: PlatformIO Zscaler setup

mharrisvr, 2021-09-23:
Hi, I'm working with a group at my work to try and get PlatformIO up and running with our Zscaler proxy. 

We are in an environment that blocks access to the 'api.registry.ns1.platformio.org' URL because of where it comes from geographically. Originally when the extension was attempted to be installed, it would fail until we unchecked the box for the proxyStrictSSL, and then the extension installed without any issues. 

This issue with blocking api.registry.ns1.platformio.org is that it causes an SSL error when trying to build, upload, upgrade, etc platformio.. now this was resolved by creating a project and running it under a different framework, in our case ESPIDF worked to get around this issue but still gives errors for accessing pypi.org and files.pythonhosted.org even though those sites are not being blocked. My thoughts are that Zscaler is replacing the SSL certificates for platformIO to access those sites, and not allowing the connection to be made, but I can't seem to find any other place we can update the settings / add in the zscaler root certificates like some other applications require to make this work. 

Here is a copy of how the settings are configured in Visual Studio Code
"workbench.colorTheme": "Default Dark+",
    "platformio-ide.buildTask": "",
    "http.proxy": "http://proxy.companyname.com:portnumber",
    "http.proxyStrictSSL": false,
    "http.proxyAuthorization": null,
    "http.proxySupport": "on",
    "http.systemCertificates": true

ivankravets, 2021-09-24:
So, and how can we resolve it? Is this a PlatformIO issue?

mharrisvr, 2021-09-27:
I was hoping someone could help point me in the direction of setting up platformIO to work offline if possible, or any other place or setting to add in proxy settings / configure platformIO/VS-code, so i don't think it's an actual issue with PlatformIO, more that we just can't seem to find additional settings / options that would allow us to configure it for use in our environment like other softwares we use.

ivankravets, 2021-09-27:
Please forward to https://community.platformio.org/c/platformio-ide/10

ivankravets, 2021-11-05:
Could someone help us with this issue?

chrisqwertz, 2024-01-19:
Hi,

for those still looking for a solution, this solved all issues for me at work (we're also using ZScaler):

Download pip.txt and rename it to pip.ini

Save the file to the following 2 directories:

- `C:\Users\%userprofile%\AppData\Roaming\PIP\pip.ini`
- `C:\Users\%userprofile%\PIP\pip.ini`

Necessary content of pip.ini:
```
[global]
trusted-host = pypi.python.org
               pypi.org
               files.pythonhosted.org
               piwheels.org
```
[pip.txt](https://github.com/platformio/platformio-vscode-ide/files/13988449/pip.txt)

mharrisvr, 2024-01-19:
Hey Chris, that is a super neat solution and I think it might be
interesting to try out next time I come across a pip configuration issue.

On our end, we discovered you could append our Zscaler.pem certificate's
contents to the cacert.pem file found within pip and it would start to work
out fine.. but we have still seen issues with some download links (Usually
if you go to them in the browser and download they will work fine after).
but your solution looks like it could possibly fix that too.

It's roughly stored somewhere around this location depending on python
version, and install directory.

C:\Users\<username>\AppData\Local\Programs\Python\PythonVersion\Lib\site-packages\pip\_vendor\certifi\cacert.pem

I'll give it a shot next time I come across it!

chrisqwertz, 2024-01-19:
That was my first approach too, it had already solved issues with another IDE (I think it was MPLAB X, but I can't be sure), but we still couldn't get the VsCode/PlatformIO combo to work. By the way: A shoutout to Knorr-Bremse's IT support, for providing the solution.

chrisqwertz, 2025-04-16:
Great, many thanks for your update regarding this issue, @lpdx !