platformio/platformio-vscode-ide #2693: Start PIO Home Server

ilpez, 2021-09-25:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.60.2
  PIO IDE: v2.3.3
  System: Darwin, 20.6.0, arm64

  %23 Exception
  ```
  Error: Could not start PIO Home server: Error: connect EADDRNOTAVAIL 127.0.0.1:8081 - Local (192.168.1.21:60644)
	at /Users/ilpez/.vscode/extensions/platformio.platformio-ide-2.3.3/node_modules/platformio-node-helpers/dist/index.js:1:5946
	at runMicrotasks (<anonymous>)
	at processTicksAndRejections (internal/process/task_queues.js:93:5)
  ```

ivankravets, 2021-11-05:
Could you reproduce this issue when restarting your OS?

ilpez, 2021-11-10:
Well it is because the cloudflare warp app on my mac, it somehow messed up the routing table and cannot even get my local react app to show up on localhost

closed 2021-11-10 by ilpez