platformio/platformio-vscode-ide #2703: PlatformIO errors while trying to install

mutcit8, 2021-10-02:
Error: TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received null at c:\Users\Newman\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771390 at a (c:\Users\Newman\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771284) at w (c:\Users\Newman\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771035) at u (c:\Users\Newman\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:770559) at c:\Users\Newman\.vscode\extensions\platformio.platformio-ide-...

Error: Could not start PIO Home server: Error: timeout at c:\Users\Newman\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:5946 at runNextTicks (internal/process/task_queues.js:58:5) at processTimers (internal/timers.js:494:9)...

MichaNoob63, 2021-10-04:
Please install the newest version of python. I had the same problem, whith newest python (3.9.7 for windows) it works! Sorry for my bad english.

MichaNoob63, 2021-10-04:
I found this tip on https://community.platformio.org/t/error-after-trying-to-install-platformio-ide-extension-in-vsc/23752

maxgerhardt, 2021-10-04:
Installing Python and Microsoft's Python versions definitely has something to do with it, someone at https://community.platformio.org/t/platformio-extension-seems-corrupt/23787/7?u=maxgerhardt observed the same

GDRPrint, 2021-10-05:
I have successfully installed Platform IO yesterday after having this issue
1 Clean uninstall VS Code instr here https://debug.to/1074/how-to-uninstall-vs-code-completely
2 Re-install VS code, close program
3 Install Python, check path update option during install, Very Important !!
4 Add Platform IO ext.
Hope this helps ! The only way that worked for me, Win 10 64 bit machine.

MichaNoob63, 2021-10-05:
@maxgerhardt 
Hallo, ja Python ist in der aktuellsten Version mit Pfad Update Option der Schlüssel! Benutze ebenfalls W10 64 bit.
Danach funktionierte die Installation von platformio ohne Probleme. Nach Anpassung in der platformio.ini konnte ich 
eine Marlinfirmware erfolgreich kompilieren. Nun geht es an die Feinheiten, leider sind manche Firmwares nur fertig kompiliert zu finden, besser wären die Quelldateien.

ivankravets, 2021-10-09:
Could someone who experiences (or manually fixed) this issue contact me via me@ikravets.com? It would be great if we find out what causes this issue.

ivankravets, 2021-10-09:
@brunoclc could you remove ALL Pythons from the system and restart VSCode? PlatformIO IDE extension comes with a built-in Python interpreter. If would be great to understand why it does not work.

closed 2021-11-05 by ivankravets