platformio/platformio-vscode-ide #2707: Error: TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received null

lixuanluo, 2021-10-04:
所
Issue Type: <b>Bug</b>

Error: TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received null
  at c:\Users\Hi\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771390
  at a (c:\Users\Hi\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771284)
  at w (c:\Users\Hi\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771035)
  at u (c:\Users\Hi\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:770559)
  at c:\Users\Hi\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771343
  at new Promise (<anonymous>)
  at l (c:\Users\Hi\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771322)
  at e.callInstallerScript (c:\Users\Hi\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:7215)
  at async H.value (c:\Users\Hi\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:763673)
  at async t.default.install (c:\Users\Hi\.vscode\extensions\platformio.platformio-ide-2.3.3\dist\extension.js:1:6773)
  at async c:\Users\Hi\.vscode\extensions\platformio.platformio-ide-2.3.3\dist\extension.js:1:12776

Extension version: 2.3.3
VS Code version: Code 1.60.2 (7f6ab5485bbc008386c4386d08766667e155244e, 2021-09-22T12:00:31.514Z)
OS version: Windows_NT x64 10.0.19043
Restricted Mode: No

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i7-7820HK CPU @ 2.90GHz (8 x 2904)|
|GPU Status|2d_canvas: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>oop_rasterization: enabled<br>opengl: enabled_on<br>rasterization: enabled<br>skia_renderer: enabled_on<br>video_decode: enabled<br>vulkan: disabled_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|31.96GB (25.16GB free)|
|Process Argv|--crash-reporter-id 5b383ed5-2db5-4f5c-99f8-775db02792c0|
|Screen Reader|no|
|VM|0%|
</details><details>
<summary>A/B Experiments</summary>

```
vsliv368:30146709
vsreu685:30147344
python383:30185418
pythonvspyt602:30300191
vspor879:30202332
vspor708:30202333
vspor363:30204092
pythonvspyt639:30300192
pythontb:30283811
pythonvspyt551:30345470
pythonptprofiler:30281270
vsdfh931:30280409
vshan820:30294714
vstes263cf:30335440
pythondataviewer:30285071
pythonvsuse255:30340121
vscod805cf:30301675
pythonvspyt200:30340761
binariesv615:30325510
vsccppwtct:30364498
pythonvssor306:30344512
bridge0708:30335490
pygetstartedt2:30371810
dockerwalkthru:30370836
bridge0723:30353136
pythonrunftest32:30373476
pythonf5test824:30373475
javagetstartedc:30364665
pythonvspyt187:30373474
pydsgst2:30361792
vsqsis300:30374797
vsaa593:30376534

```

</details>

SyntaxFrasse, 2021-10-05:
Me too. Tryied a clean install of visual studio code on 2 different computers with windows 10 x64 and 1 windows 7 new installation x64 and have the same problem on all of them.

lixuanluo, 2021-10-05:
Yep.But I was fine on Mac, and Windows kept reporting errors

robotikman, 2021-10-05:
I'm getting this error too, how can we fix it ?? can you help me ??

robotikman, 2021-10-05:
I am also getting this error, how can we fix it ?? can you help me ??

lixuanluo, 2021-10-05:
I can't fix it，I will use Arduino IDE

robotikman, 2021-10-05:
Yes i couldn't fix it no matter what i do but i want to use this program.

lixuanluo, 2021-10-05:
Maybe the next version will fix it.

I'm going to use arduinoIDE before that

SyntaxFrasse, 2021-10-05:
I tryied installing Python, but it did not help. Also tryied the other solutions, but nothing helped.

robotikman, 2021-10-05:
Yes, I tried them too, but unfortunately it did not bring a solution.

GDRPrint, 2021-10-05:
I have successfully installed Platform IO yesterday after having this issue
1 Clean uninstall VS Code   instr here  https://debug.to/1074/how-to-uninstall-vs-code-completely
2 Re-install VS code, close program
3 Install Python, check path update option during install, Very  Important !!
4 Add Platform IO ext.
Hope this helps ! The only way that worked for me, Win 10 64 bit machine.

lixuanluo, 2021-10-05:
I am very frustrated. After I completely deleted the VS Code &amp;Python environment and installed it directly, I still got an error. What version of Python do you have installed?Was it installed in VS Code?

lixuanluo, 2021-10-05:
it's too difficult

ejikpolemidis, 2021-10-05:
> I have successfully installed Platform IO yesterday after having this issue 1 Clean uninstall VS Code instr here https://debug.to/1074/how-to-uninstall-vs-code-completely 2 Re-install VS code, close [...]

first install previus version and after update to newest.this worked for me

robotikman, 2021-10-05:
> > Platform IO'yu dün bu sorunu yaşadıktan sonra başarıyla yükledim 1 VS Code instr'yi buradan temizleyin https://debug.to/1074/how-to-uninstall-vs-code-completely 2 VS kodunu yeniden yükleyin, programı kapatın 3 Install Python, kurulum sırasında yol güncelleme seçeneğini kontrol edin, Çok Önemli !! 4 Platform IO dahili ekleyin. Bu yardımcı olur umarım ! Benim için çalışan tek yol, Win 10 64 bit makine.
> 
> ilk önce önceki sürümü yükleyin ve en yenisine güncelledikten sonra bu benim için çalıştı

Installing the previous version also does not work.

lixuanluo, 2021-10-05:
> It doesn't work, other solutions are needed.
>  Failed to install PlatformIO IDE.

yep.I will use Arduino IDE，Platform IO is unbelievable

lixuanluo, 2021-10-05:
> evet.Arduino IDE kullanacağım,Platform IO inanılmaz
>  …
>
> Working on Arduino IDE is a bit boring.
>  Visual code seems easier to work with.
>  I wish we could upload and work from here.
>  We would be very happy if those who know the solution share.

I hope developers can solve this problem in the next version.

GDRPrint, 2021-10-06:
> I am very frustrated. After I completely deleted the VS Code &amp;Python environment and installed it directly, I still got an error. What version of Python do you have installed?Was it installed in VS Code?
> […](#)

No I installed Python as a separate entity, remember my remark, Close VS Code. I installed the latest version of Python. I then closed Python, opened VS Code and added the Platform IO extension. sorry if this does not work for  others. I too was very frustrated until I found this solution.

GDRPrint, 2021-10-06:
> I just installed an earlier version ( 1yr old) which prompted me to install Python and tick the 'PATH' option on install. Then restarted VSCode, worked OK , so updated to the latest version and it worked

I tried installing VS code from Nov 2020, but had no success. I did not know then about the method I describe above, so am guessing it was bound to fail.

borisgraell, 2021-10-06:
> It doesn't work, other solutions are needed. Failed to install PlatformIO IDE.

I solved this problem reinstalling python and clicking the "add to PATH" option while installing .

SyntaxFrasse, 2021-10-06:
> first install previus version and after update to newest.this worked for me

Where is the "Check path" function when installing Python? I don't get that option.
I am downloading python directly from here https://www.python.org/downloads/ and i am using the windows installer.

lixuanluo, 2021-10-06:
great！it's helpful for me！
thanks bro

     

  
  
  
right at the beginning 
  
AAAAARGHHHH, how could i miss that. I am so sorry..... Now platformio works here too.
 Just started the Marlin build for my 3D printer.
 
Thanx dude. If i ever pass by i will buy you a huge beer.!!!!

manicresin, 2021-10-06:
I just had this same issue and installing Python with the add to path checked fixed it, thank you.

robotikman, 2021-10-06:
I had installed python before but had trouble uninstalling it.
I kept getting different errors when trying to uninstall python.
I installed other programs to fix the errors.
The computer was full, overflowing, and thoroughly confused.
I got angry and formatted the computer from scratch.
I reinstalled Windows.
I installed Python by marking patch 3.10 as you said.
Then I installed the visual studio code.
Then I installed the IDE platform and I didn't get any errors, thank goodness we reached the result.
I'm happy now and thank you all for your help :))

robotikman, 2021-10-06:
I had installed python before but had trouble uninstalling it.
I kept getting different errors when trying to uninstall python.
I installed other programs to fix the errors.
The computer was full, overflowing, and thoroughly confused.
I got angry and formatted the computer from scratch.
I reinstalled Windows.
I installed Python by marking 3.10 patch as you said.
Then I installed the visual studio code.
Then I installed Platform IDE through the program and I didn't get any errors, thank goodness we got what we wanted.
I'm happy now and thank you all for your help :))

robotikman, 2021-10-06:
I had installed python before but had trouble uninstalling it.
I kept getting different errors when trying to uninstall python.
I installed other programs to fix the errors.
The computer was full, overflowing, and thoroughly confused.
I got angry and formatted the computer from scratch.
I reinstalled Windows.
I installed Python by marking 3.10 patch as you said.
Then I installed the visual studio code.
Then I downloaded and installed Platform IDE with visual studio code.
I didn't get any errors.
Thankfully, we finally got what we wanted.
I'm happy now and thank you all for your help :))

robotikman, 2021-10-09:
> Birisi bu sorunu yeniden oluşturmak için uzak bir oturum sağlayabilir mi? Çok nadir görülen bir sorun gibi görünüyor ve sistem yapılandırmanızla bağlantılı. [Ayrıntılar](mailto:me@ikravets.com) için lütfen [me@ikravets.com](mailto:me@ikravets.com) adresine e-posta [gönderin](mailto:me@ikravets.com) .

No. There is no such thing as a rare encounter with such a problem.
There may be others in the future.
Anyone encountering such a problem can easily find a solution here.

ivankravets, 2021-10-09:
PlatformIO IDE extension comes with built-in Python. You don't need to install an external Python interpreter. If built-in Python does not work, need to investigate what is going wrong.

robotikman, 2021-10-09:
> PlatformIO IDE uzantısı yerleşik Python ile birlikte gelir. Harici bir Python yorumlayıcısı yüklemeniz gerekmez. Yerleşik Python çalışmıyorsa, neyin yanlış gittiğini araştırmanız gerekir.

If so why did we encounter such a problem ??
The issue has been resolved and this bug report will stay here.

ivankravets, 2021-10-09:
@robotikman yes, the sun is shining in the sky.

robotikman, 2021-10-09:
@ivankravets yes thank you sir.
Let the sun light your way.

bjanders, 2021-10-10:
I had this problem and it was fixed by installing Python and adding Python to the path. However, the installation instructions don't even mention that Python should be installed first. The install instructions on https://platformio.org/install/ide?install=vscode says:

1. Open VSCode Extension Manager
2. Search for official PlatformIO IDE extension
3. Install PlatformIO IDE.

The page should amend that you need to make sure that Python 3 is installed and that the command is in your path before proceeding with the extension installation. Even if this is mentioned on some other page, it should be repeated here for clarity.

closed 2021-11-05 by ivankravets