platformio/platformio-vscode-ide #2712: PlatformIO Marlin [Errno 2]

GurkeX, 2021-10-06:
C:\Users\rk\AppData\Local\Programs\Python\Python310\python.exe: can't open file 'C:\\Users\\rk\\.platformio\\packages\\tool-scons\\scons.py': [Errno 2] No such file or directory

Hi I have recently updated Python to the newest version and since then i´m getting this error and can´t build marlin anymore has anyone got experience with this problem and may be able to help out?

ivankravets, 2021-11-05:
Please update VSCode to the latest 1.62.0 or above version, and install the latest (2.4.0 or above) PlatformIO IDE extension. See https://marketplace.visualstudio.com/items/platformio.platformio-ide/changelog

Does it work now?

closed 2021-11-05 by ivankravets

mattdog01, 2021-11-07:
Hello I have not been in VSCode for a little while and I'm having this issue now. I have updated python to the latest, updated VSCode to the latest and the platformIO IDE extension is v2.4.0. any thoughts why it is still happening?