platformio/platformio-vscode-ide #2778: Won't launch

fabien85100, 2021-11-01:
Please do smothing about platformio, we are a lot of people having issues with it and for many 3d printig mother board we only can use plateformio.
After 1 day of use PlatformiO won't start up. i try to unistall it and do a fresh install but it still not work....

ivankravets, 2021-11-01:
Could you provide more details on what is going wrong?

fabien85100, 2021-11-02:
i find how to resolve  my problem, i uninstall vs code and remove all folder: ".platformio" ".vscode" and "code" (this one is in appdata) . And i let python install.
Thank you.

Bye.

ivankravets, 2021-11-02:
Did you install any Python versions before?

fabien85100, 2021-11-02:
yes and it was not working at all...
That's why i remove everything (exept python) and reinstall all and now it work like a charm !

closed 2021-11-05 by ivankravets