platformio/platformio-vscode-ide #2788: Start PIO Home Server

OldThank7, 2021-11-03:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.61.2
  PIO IDE: v2.3.4
  System: Windows_NT, 10.0.22000, x64

  %23 Exception
  ```
  Error: C:\Users\OldThank\.platformio\python3\python.exe: No module named platformio

	at c:\Users\OldThank\.vscode\extensions\platformio.platformio-ide-2.3.4\node_modules\platformio-node-helpers\dist\index.js:1:5850
	at ChildProcess.a (c:\Users\OldThank\.vscode\extensions\platformio.platformio-ide-2.3.4\node_modules\platformio-node-helpers\dist\index.js:1:771396)
	at ChildProcess.emit (events.js:315:20)
	at ChildProcess.EventEmitter.emit (domain.js:467:12)
	at ChildProcess.cp.emit (c:\Users\OldThank\.vscode\extensions\platformio.platformio-ide-2.3.4\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1048:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
  ```

ivankravets, 2021-11-03:
Can you reproduce this issue when restarting VSCode? Did you uncheck "Use built-in PlatformIO Core" in VSCode settings?

J-Brk, 2021-11-03:
Same issue here; tried unchecking "Use built-in PlatformIO Core" but doesn't make any difference

closed 2021-11-05 by ivankravets