platformio/platformio-vscode-ide #2797: After the update, the menu Platform item in the project tasks was lost.

SinglWolf, 2021-11-06:
In my project I use two boards: `esp-wrover-kit` and `esp32-s2-saola-1`.
After upgrading `PlatformIO` to version `2.4.0`, the `Platform` menu item disappeared in the project tasks for the `esp-wrover-kit` board.
`platfirmio.ini` for `esp32-s2-saola-1`:
```
[platformio]
src_dir = main
include_dir = main/include
description = controlPanel

[env:esp32dev]
platform = espressif32
framework = espidf
; board = esp-wrover-kit
board = esp32-s2-saola-1
monitor_speed = 115200
monitor_filters = direct
```
[image]

It's okay.

`platfirmio.ini` for `esp-wrover-kit`:
```
[platformio]
src_dir = main
include_dir = main/include
description = controlPanel

[env:esp32dev]
platform = espressif32
framework = espidf
board = esp-wrover-kit
; board = esp32-s2-saola-1
monitor_speed = 115200
monitor_filters = direct
```
[image]
`Platform` menu item not.
Removing `PlatformIO` and reinstalling from scratch (with removing the `.platformio` directory) does not help.
What have I done wrong?

ivankravets, 2021-11-06:
Sorry,  I can't reproduce this issue. Could you try to restart VSCode?
[image]

SinglWolf, 2021-11-06:
> Sorry, I can't reproduce this issue. Could you try to restart VSCode?

I even rebooted the computer.
You showed me what works for me.
Look at the picture above.
No menu if board `esp-wrover-kit`

ivankravets, 2021-11-06:
Could you try this example https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-blink ?

[image]

SinglWolf, 2021-11-06:
> Could you try this example https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-blink ?

Strange. OK. Once I check it out, I'll be back. A little busy.

SinglWolf, 2021-11-06:
> Could you try this example https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-blink ?

Does not work.
[image]
Menu item `Platform` appears only when board `esp32-s2-saola-1` is selected.
I have `Windows 10 Pro` OS. Python version `3.9.2`.

SinglWolf, 2021-11-06:
When I try to compile the firmware, I get this:
```
> Executing task: C:\Users\Alexander\.platformio\penv\Scripts\platformio.exe run --environment esp32dev <

Processing esp32dev (platform: espressif32; framework: espidf; board: esp32dev)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.3.2+sha.824c5fb) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 3.40301.0 (4.3.1)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-ninja 1.9.0
 - toolchain-esp32ulp 1.22851.191205 (2.28.51)
TypeError: stat: path should be string, bytes, os.PathLike or integer, not NoneType:
  File "C:\Users\Alexander\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 180:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Alexander\.platformio\packages\tool-scons\scons-local-4.2.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Alexander\.platformio\packages\tool-scons\scons-local-4.2.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Alexander\.platformio\platforms\espressif32\builder\main.py", line 233:
    target_elf = env.BuildProgram()
  File "C:\Users\Alexander\.platformio\packages\tool-scons\scons-local-4.2.0\SCons\Util.py", line 748:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Alexander\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 61:    
    env.ProcessProgramDeps()
  File "C:\Users\Alexander\.platformio\packages\tool-scons\scons-local-4.2.0\SCons\Util.py", line 748:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Alexander\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 128:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\Alexander\.platformio\packages\tool-scons\scons-local-4.2.0\SCons\Util.py", line 748:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Alexander\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 343:
    SConscript(env.GetFrameworkScript(f), exports="env")
  File "C:\Users\Alexander\.platformio\packages\tool-scons\scons-local-4.2.0\SCons\Script\SConscript.py", line 660:
    return method(*args, **kw)
  File "C:\Users\Alexander\.platformio\packages\tool-scons\scons-local-4.2.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Alexander\.platformio\packages\tool-scons\scons-local-4.2.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Alexander\.platformio\platforms\espressif32\builder\frameworks\espidf.py", line 63:
    assert os.path.isdir(TOOLCHAIN_DIR)
  File "C:\Users\Alexander\.platformio\python3\lib\genericpath.py", line 42:
    st = os.stat(s)
```

ivankravets, 2021-11-06:
> Does not work.

Do you see "indicator" on the bottom status bar? "indexing..."? Could you build your project?

SinglWolf, 2021-11-06:
> Could you build your project?

When I try to create a new project, I get the following:
```
PIO Core Call Error: "The current working directory D:\\WORK\\ESP-IDF\\TEST1 will be used for the project.\r\n\r\nThe next files/directories have been created in D:\\WORK\\ESP-IDF\\TEST1\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Processing esp32dev (platform: espressif32; board: esp32dev; framework: espidf)\r\n--------------------------------------------------------------------------------\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nCONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html\r\nPLATFORM: Espressif 32 (3.3.2+sha.824c5fb) > Espressif ESP32 Dev Module\r\nHARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash\r\nDEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)\r\nPACKAGES: \r\n - framework-espidf 3.40301.0 (4.3.1) \r\n - tool-cmake 3.16.4 \r\n - tool-esptoolpy 1.30100.210531 (3.1.0) \r\n - tool-idf 1.0.1 \r\n - tool-mconf 1.4060000.20190628 (406.0.0) \r\n - tool-ninja 1.9.0 \r\n - toolchain-esp32ulp 1.22851.191205 (2.28.51)\r\nTypeError: stat: path should be string, bytes, os.PathLike or integer, not NoneType:\r\n File \"C:\\Users\\Alexander\\.platformio\\penv\\Lib\\site-packages\\platformio\\builder\\main.py\", line 180:\r\n env.SConscript(\"$BUILD_SCRIPT\")\r\n File \"C:\\Users\\Alexander\\.platformio\\packages\\tool-scons\\scons-local-4.2.0\\SCons\\Script\\SConscript.py\", line 597:\r\n return _SConscript(self.fs, *files, **subst_kw)\r\n File \"C:\\Users\\Alexander\\.platformio\\packages\\tool-scons\\scons-local-4.2.0\\SCons\\Script\\SConscript.py\", line 285:\r\n exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)\r\n File \"C:\\Users\\Alexander\\.platformio\\platforms\\espressif32\\builder\\main.py\", line 233:\r\n target_elf = env.BuildProgram()\r\n File \"C:\\Users\\Alexander\\.platformio\\packages\\tool-scons\\scons-local-4.2.0\\SCons\\Util.py\", line 748:\r\n return self.method(*nargs, **kwargs)\r\n File \"C:\\Users\\Alexander\\.platformio\\penv\\Lib\\site-packages\\platformio\\builder\\tools\\platformio.py\", line 61:\r\n env.ProcessProgramDeps()\r\n File \"C:\\Users\\Alexander\\.platformio\\packages\\tool-scons\\scons-local-4.2.0\\SCons\\Util.py\", line 748:\r\n return self.method(*nargs, **kwargs)\r\n File \"C:\\Users\\Alexander\\.platformio\\penv\\Lib\\site-packages\\platformio\\builder\\tools\\platformio.py\", line 128:\r\n env.BuildFrameworks(env.get(\"PIOFRAMEWORK\"))\r\n File \"C:\\Users\\Alexander\\.platformio\\packages\\tool-scons\\scons-local-4.2.0\\SCons\\Util.py\", line 748:\r\n return self.method(*nargs, **kwargs)\r\n File \"C:\\Users\\Alexander\\.platformio\\penv\\Lib\\site-packages\\platformio\\builder\\tools\\platformio.py\", line 343:\r\n SConscript(env.GetFrameworkScript(f), exports=\"env\")\r\n File \"C:\\Users\\Alexander\\.platformio\\packages\\tool-scons\\scons-local-4.2.0\\SCons\\Script\\SConscript.py\", line 660:\r\n return method(*args, **kw)\r\n File \"C:\\Users\\Alexander\\.platformio\\packages\\tool-scons\\scons-local-4.2.0\\SCons\\Script\\SConscript.py\", line 597:\r\n return _SConscript(self.fs, *files, **subst_kw)\r\n File \"C:\\Users\\Alexander\\.platformio\\packages\\tool-scons\\scons-local-4.2.0\\SCons\\Script\\SConscript.py\", line 285:\r\n exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)\r\n File \"C:\\Users\\Alexander\\.platformio\\platforms\\espressif32\\builder\\frameworks\\espidf.py\", line 63:\r\n assert os.path.isdir(TOOLCHAIN_DIR)\r\n File \"C:\\Users\\Alexander\\.platformio\\python3\\lib\\genericpath.py\", line 42:\r\n st = os.stat(s)\r\n========================== [FAILED] Took 0.41 seconds =========================="
```