platformio/platformio-vscode-ide #2811: Start PIO Home Server

Andradeerik, 2021-11-18:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.62.3
  PIO IDE: v2.4.0
  System: Windows_NT, 10.0.22000, x64

  %23 Exception
  ```
  Error: Error: Traceback (most recent call last):
  File "C:\Users\dani_\.platformio\penv\lib\site-packages\platformio\__main__.py", line 129, in main
    cli()  %23 pylint: disable=no-value-for-parameter
  File "C:\Users\dani_\.platformio\penv\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\dani_\.platformio\penv\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "C:\Users\dani_\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "C:\Users\dani_\.platformio\penv\lib\site-packages\click\core.py", line 1256, in invoke
    Command.invoke(self, ctx)
  File "C:\Users\dani_\.platformio\penv\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\dani_\.platformio\penv\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "C:\Users\dani_\.platformio\penv\lib\site-packages\click\decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "C:\Users\dani_\.platformio\penv\lib\site-packages\platformio\__main__.py", line 67, in cli
    maintenance.on_platformio_start(ctx, force, caller)
  File "C:\Users\dani_\.platformio\penv\lib\site-packages\platformio\maintenance.py", line 49, in on_platformio_start
    after_upgrade(ctx)
  File "C:\Users\dani_\.platformio\penv\lib\site-packages\platformio\maintenance.py", line 172, in after_upgrade
    update_core_packages(silent=True)
  File "C:\Users\dani_\.platformio\penv\lib\site-packages\platformio\package\manager\core.py", line 64, in update_core_packages
    if not silent or pm.outdated(pkg, spec).is_outdated():
  File "C:\Users\dani_\.platformio\penv\lib\site-packages\platformio\package\manager\_update.py", line 49, in outdated
    reg_pkg = self.fetch_registry_package(pkg.metadata.spec)
  File "C:\Users\dani_\.platformio\penv\lib\site-packages\platformio\package\manager\_registry.py", line 144, in fetch_registry_package
    result = regclient.get_package(self.pkg_type, spec.owner, spec.name)
  File "C:\Users\dani_\.platformio\penv\lib\site-packages\platformio\clients\registry.py", line 125, in get_package
    return self.fetch_json_data(
  File "C:\Users\dani_\.platformio\penv\lib\site-packages\platformio\clients\http.py", line 143, in fetch_json_data
    with ContentCache("http") as cc:
  File "C:\Users\dani_\.platformio\penv\lib\site-packages\platformio\cache.py", line 36, in __enter__
    self.delete()
  File "C:\Users\dani_\.platformio\penv\lib\site-packages\platformio\cache.py", line 132, in delete
    with open(self._db_path, mode="w", encoding="utf8") as fp:
PermissionError: [Errno 13] Permission denied: 'C:\\Users\\dani_\\.platformio\\.cache\\http\\db.data'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

    at cmd (c:\Users\dani_\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\home.js:195:11)
    at ChildProcess.r (c:\Users\dani_\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:177:7)
    at ChildProcess.emit (events.js:315:20)
    at ChildProcess.EventEmitter.emit (domain.js:467:12)
    at ChildProcess.cp.emit (c:\Users\dani_\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\cross-spawn\lib\enoent.js:34:29)
    at maybeClose (internal/child_process.js:1048:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
  ```

Andradeerik, 2021-11-20:
This worked for me
delete folder ".platformio"
in my case it is in  "C:\Users\dani_\.platformio\"
uninstall platformio extension
restart vscode
install platformio extension
and everything returned to normal

closed 2021-11-20 by Andradeerik

ear9mrn, 2021-12-01:
didn't fix my problem. Running on Ubuntu.