platformio/platformio-vscode-ide #2839: Start PIO Home Server

laurenthugues, 2021-12-13:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.63.0
  PIO IDE: v2.4.0
  System: Darwin, 21.1.0, arm64

  %23 Exception
  ```
  Error: Error: You are not connected to the Internet.
PlatformIO needs the Internet connection to download dependent packages or to work with PlatformIO Account.

	at /Users/laurent/.vscode/extensions/platformio.platformio-ide-2.4.0/node_modules/platformio-node-helpers/dist/index.js:1:9851
	at ChildProcess.r (/Users/laurent/.vscode/extensions/platformio.platformio-ide-2.4.0/node_modules/platformio-node-helpers/dist/index.js:1:5134)
	at ChildProcess.emit (events.js:315:20)
	at ChildProcess.EventEmitter.emit (domain.js:467:12)
	at maybeClose (internal/child_process.js:1048:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
  ```

ivankravets, 2022-02-05:
> Error: Error: You are not connected to the Internet.
> PlatformIO needs the Internet connection to download dependent packages or to work with PlatformIO Account.

Please check your internet connection and restart VSCode.

Does it work now?

closed 2022-02-05 by ivankravets

laurenthugues, 2022-03-05:
Hello,
I am connected to internet. I found a workaround by commenting the lines below in the file /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/platformio/clients/http.py

#s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
#s.connect((host, 80))
#s.close()