platformio/platformio-vscode-ide #2850: Start PIO Home Server

tungthanhbui, 2021-12-21:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.63.2
  PIO IDE: v2.4.0
  System: Linux, 5.11.0-43-generic, x64

  %23 Exception
  ```
  Error: Usage: python -m platformio [OPTIONS] COMMAND [ARGS]...
Try 'python -m platformio -h' for help.

Error: No such command "home"

	at /home/tung_thanh_bui/.vscode/extensions/platformio.platformio-ide-2.4.0/node_modules/platformio-node-helpers/dist/index.js:1:9851
	at ChildProcess.r (/home/tung_thanh_bui/.vscode/extensions/platformio.platformio-ide-2.4.0/node_modules/platformio-node-helpers/dist/index.js:1:5134)
	at ChildProcess.emit (events.js:315:20)
	at ChildProcess.EventEmitter.emit (domain.js:467:12)
	at maybeClose (internal/child_process.js:1048:16)
	at Socket.<anonymous> (internal/child_process.js:439:11)
	at Socket.emit (events.js:315:20)
	at Socket.EventEmitter.emit (domain.js:467:12)
	at Pipe.<anonymous> (net.js:673:12)
  ```

ear9mrn, 2022-01-06:
[https://github.com/platformio/platformio-vscode-ide/issues/2859#issue-1090767446](url)

ivankravets, 2022-02-05:
Have you resolved this issue? Have you had PlatformIO installed before?

closed 2022-02-11 by ivankravets