platformio/platformio-vscode-ide #2866: Start PIO Home Server

dufduf-one, 2022-01-05:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues

  %23 Configuration

  VSCode: 1.63.2
  PIO IDE: v2.0.9
  System: Windows_NT, 10.0.19042, x64

  %23 Exception
  ```
  Error: Usage: platformio [OPTIONS] COMMAND [ARGS]...
Try 'platformio -h' for help.

Error: No such command "home"

    at stderr (c:\Users\franck dufeil\.vscode\extensions\comp2300-anu.comp2300-platformio-ide-2.0.9\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\home.js:197:9)
    at ChildProcess.d (c:\Users\franck dufeil\.vscode\extensions\comp2300-anu.comp2300-platformio-ide-2.0.9\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\misc.js:297:62)
    at ChildProcess.emit (events.js:315:20)
    at ChildProcess.EventEmitter.emit (domain.js:467:12)
    at ChildProcess.cp.emit (c:\Users\franck dufeil\.vscode\extensions\comp2300-anu.comp2300-platformio-ide-2.0.9\node_modules\cross-spawn\lib\enoent.js:34:29)
    at maybeClose (internal/child_process.js:1048:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
  ```

ear9mrn, 2022-01-06:
[https://github.com/platformio/platformio-vscode-ide/issues/2859#issue-1090767446](url)

ivankravets, 2022-02-05:
Have you resolved this issue? Have you had PlatformIO installed before?

closed 2022-02-11 by ivankravets